slic3r/Slic3r #4208: Mouse wheel zoom should zoom centered wherever the cursor is

VanessaE, 2017-11-28:
### Version
commit 202a90ff903be5b61fca317c6b614af0d5f34d8a

### Operating system type + version
Debian 9

### Behavior

Most programs that can zoom with the mouse wheel do so with the center of the zoom action at the mouse pointer, while Slic3r always centers the zoom on the center of the viewport.  This makes one need extra drag actions to zoom in on one object or part of the plate.  Compare to how zoom works on gcode.ws or in Blender.

#### STL/Config (.ZIP) where problem occurs

None needed.

Keavon, 2017-12-03:
I oppose this (at least by default). From my experience, this is a very annoying feature because it doesn't give control over where to zoom. Zooming in and out is reproducible if it keeps the same center point, and intuitively it should go towards the center of the viewport, not somewhere dependant upon an always changing value such as mouse position. Blender is a very common 3D package used widely by the same community and it does not do this by default.

VanessaE, 2017-12-04:
The GIMP and Pronterface also do it the way I suggest.

lordofhyphens, 2017-12-04:
So does Simulink, but that doesn't necessarily make it a good default. 

At minimum I think this would need to be a selectable behavior.