slic3r/Slic3r #4209: Slic3r should not invoke travel-lift-Z (if enabled) at layer change if next start X/Y is similar to previous endpoint

VanessaE, 2017-11-29:
### Version
commit 202a90ff903be5b61fca317c6b614af0d5f34d8a

### Operating system type + version

Debian 9

### Behavior

Title says it all

#### STL/Config (.ZIP) where problem occurs
[Retraction.stl.zip](https://github.com/alexrj/Slic3r/files/1512065/Retraction.stl.zip)

No config needed

Tinchus2009, 2017-12-11:
on the contrary. IF you dont use retrarc in Z, your seam probably will be worst if your extruder is leaking. I have experienced you situation, specially for example when printing a calibration extrusion box. But looking at that seem is what tells me , for example, if my PET has taken humidity, I can tell looking how much is leaking. If I get rid of the Z lifting, the seam is way worst

VanessaE, 2017-12-11:
Retract != z-lift.  I'm suggesting discontinuing the latter on layer change (that is, Z is already being raised anyway) if the nozzle is already within a reasonable range of the seam.  Z-lift right at the layer change invokes an up-down-up move pattern which wastes time (making any oozing problem you describe even worse).