slic3r/Slic3r #4210: Slic3r 3d View Not displaying correctly.

irvcm, 2017-11-30:
### Version
1.2.9

### Operating system type + version
Windows 10 x64 

### Behavior
Slic3r Opens and the 3d views comes up as all white. Appears to be the same issue as Case #3654, I've loaded a model and can see it in 2d view but the 3d view appears all white. Not an issue in Slic3r Prusa

lordofhyphens, 2017-11-30:
Confirm that this is an issue on 1.3.0-dev, please.

irvcm, 2017-12-06:
Its still and issue in the 1.3.0 dev version, Same results.

lordofhyphens, 2017-12-07:
Going to need to know more about your system, including video drivers, card
make, etc.

Whatever your problem is is OpenGL related.

lordofhyphens, 2017-12-07:
I cannot reproduce on my own laptop.

irvcm, 2017-12-14:
Ok with more testing if found out more info.

Currently the computer (Dell 3520) running Windows 10 x64 is connected to D3100 Dock. Connected to the dock are 2 HPZR2440w monitors. The issue described above only happens when Slic3r is viewed on one of the monitors. If the windows is dragged to the laptop display it functions normally. 

More computer specs (gathered from device manager)

Nvidia Quadro M620 Running driver version 22.21.13.866
Intel(R) HD Graphics  running version 22.20.16.4799
Dell D3100 usb dock running version 8.3.2008.0

irvcm, 2017-12-14:
Tested monitor without dock and it worked so I believe the problem lies with the connected dock, Tried another dell dock WD15(i think) and it worked. The original dock communicated via USB while the new dock communicated via USB C.