slic3r/Slic3r #4211: Implement retraction override in filament settings

expertmm, 2017-12-01:
### Version
1.3.0-dev (git ~Nov 29 2017) 
commit unknown since Antergos cleared my pamac-build-tmp

### Operating system type + version
Antergos (~Nov 29 2017 [antergos is rolling])

### Behavior
* filaments differ in properties under heat (for example, PET-G needs different retraction)

lordofhyphens, 2017-12-01:
One immediate workaround you could try is to use firmware retraction and
insert a command to change retraction in custom gcode.

expertmm, 2017-12-01:
If I set use firmware retraction in Print Settings, General, can I then insert the command into the Filament Settings' custom gcode?

Also, while we're here...
There are some other distinctly filament-specific settings in "Printer Settings" that should probably be moved to or overridden by Filament Settings (for example, Hatchbox PET-G seems to compress in the nozzle, then dribbles during travel, then needs a big push to get going again the longer you wait especially on tool change for multi-extruder setups): 
"Length" and "Extra length on restart"  (under "Retraction when tool is disabled...")

lordofhyphens, 2017-12-04:
Yes.

closed 2018-04-14 by lordofhyphens