slic3r/Slic3r #4212: Severe overextrusion?

Tinchus2009, 2017-12-01:
### Version
1.2.9-1250-g8ca6171

### Operating system type + version
win10 64b

### Behavior
I have been printing and have this issue with some models. Sympthom is overextrusion. First of all, my printer extrusion is really well calibrated, so the issue is not there.
But the problem is overextrusion, in some situations I can see my noozle "swimming into the plastic" , I even though I could have a Z issue... may be my Z axis was haveing a problem? I double checked, and answer is no, My Z axis is OK.
I have uploaded Gcode, image and config of my ast print (10 minutes ago) wich sohwed this problem.
Looking at the printed model, the overextrusion was in the infill, perimeters were ok.
Looking at the 3d simulation in slic3r it clearly shows the problem: the infill lines, if they are really showing the extrusion amount, are line 2.X time biggers than the perimeters (actually, each infill line is 2.4 times wider than a perimeter line), so haveing manually set my perimters extrusion to 0.48, this means that the infill lines are like almost 1 mm wide????!!!!!!!
That is wrong, my noozle is 0.4, extrusion width should not be bigger than  0.68 (I have manually set my perimeters width, but have on auto my extrusion width)
So this is the problem, for some reason, in auto mode, extrusion width sometimes get bigger than it should related to the noozle size.
[falla sobreextrusion.zip](https://github.com/alexrj/Slic3r/files/1520410/falla.sobreextrusion.zip)

lordofhyphens, 2017-12-01:
Auto width has some known failings when it comes to certain layer and
nozzle heights.

I have put some patches in to limit widths, limiting it more is another
hack that we can try, but really need an algorithm or set of algorithms
that work over different flow amounts.

bubnikv, 2017-12-01:
These are our current defaults, which are very simple, but reliable.
https://github.com/prusa3d/Slic3r/blob/master/xs/src/libslic3r/Flow.cpp

lordofhyphens, 2018-01-06:
@Tinchus2009 try that tweak; I've reduced the maximum that slic3r's auto width is allowed to go to to 140% for infill and 125% for everything else.

Tinchus2009, 2018-01-11:
If you tell me how to use this I will jejeje

lordofhyphens, 2018-01-11:
Download a current version of 1.3.0-dev.

fredizzimo, 2018-08-25:
@lordofhyphens,

I think you reversed the values, since I get 
```
; external perimeters extrusion width = 0.33mm (2.30mm^3/s)
; perimeters extrusion width = 0.40mm (2.83mm^3/s)
; infill extrusion width = 0.38mm (3.32mm^3/s)
; solid infill extrusion width = 0.40mm (3.53mm^3/s)
; top infill extrusion width = 0.40mm (2.83mm^3/s)
```

With a 0.3mm nozzle and 0.2mm layer height. 

0.3 * 1.25 = 0.38 (actually 0.375)
0*3 * 1.4 = 0.40 (these are not capped to the maximum, and follows the width formula, which gives a value close to 0.4)