slic3r/Slic3r #4213: [Feature request] SVG Export infill too

gege2b, 2017-12-01:
_Is this a new feature request?_
yup

It's about SVG Export. Actually, all it does is exporting the perimeter only

I would love to be able to export infill in SVG (I built a 2D drawing machine and would love to see it drawing honeycomb-filled shapes or whatever)

For the same reason, It would be nice if we can choose only one layer to export, but it's not a top priority, Slic3r being a 3D Slicer after all (one can deal with inkscape or whatever to delete unwanted layers after export)

lordofhyphens, 2018-01-06:
This would also be handy for SLA.

HCans, 2018-09-29:
Being able to export the infill patterns as well, would be great for laser AM processes

lordofhyphens, 2018-09-29:
Export SVG from the DLP plater.