slic3r/Slic3r #4214: xy_compensation lower than -0.1

Lebensgefahr, 2017-12-03:
### Version
1.3.0-dev

### Operating system type + version
Win7 x64

### Behavior
Iam trying to use xy_compensation lower than -0.1 but it has no effect. If i try 0.1, 0.2 or -0.1 it works, but lower than -0.1 see as xy_compensation is disabled. 
I try to get hole exactly 6.5 mm. If i used xy_compenastion +0.4 than i get 5.3 mm but if i use -0.4 i get the same result if i use -0.1.

Lebensgefahr, 2017-12-05:
My fault guys. In this version (it shipped with last version of repetier host) didnt work any xy_size_compensation below zero. But in 
Slic3r-master.2017.11.07.1542.8ca6171.64bit.zip it really works. I will use it.

closed 2017-12-05 by Lebensgefahr

lordofhyphens, 2017-12-05:
Glad to hear it.

Lebensgefahr, 2017-12-05:
It was a bug in earlier versions. It was fixed with adding abs(xy_size_compensation) >0 instead of 
 xy_size_compensation >0 in if  condition in slic3r.cpp. Printed things has a jitter about the  real size from print to print and this let me think that -0.1 works but it doesnt. And the same version did not get me the idea that my version is not the latest. 
Sorry, i will do my best next time :-)