slic3r/Slic3r #4216: "Not show again" for "Outside print bed"-Dialog

dartrax, 2017-12-05:
### Version
1.3.0 dev 2017-11-07

### Operating system type + version
Win 7 64bit

### Behavior
Every time I add a .stl to slicer (via Add button or by Drag & Drop) instantly a Dialog appears that I have to click away. It's Title is "Outside print bed".
Setting "Auto-center parts" is off.

I would expect to have the opportunity to arrange the part at least once before such a dialog occurs. It would be great if this dialog only pops up when starting G-Code generation.

GilesBathgate, 2017-12-08:
@dartrax This behaviour was introduced by my PR: https://github.com/alexrj/Slic3r/pull/4163 . I am trying to think of the most appropriate fix. I am not sure where in the code there is an event when G-Code generation starts, but I imagine that because it starts automatically (in the background) it would be just as obtrusive to present the message then. Would it be better perhaps to just place the warning in the status bar?  There is also a warning about the size of the object being too large, which was there before my change, and is equally obtrusive.

dartrax, 2017-12-12:
@GilesBathgate Hi, thanks for the explanation.
"starting G-Code generation" also may not be the best time to show the warning. My idea is, that the warning should occur at the latest possible point, so the user has time to arrange the objects before the warning appears. When the user attempts to start the print or save the G-code and there is still something outside the print bed (or the object too large), than a warning should appear.

If that doesn't fit well to the way slic3r is coded internally, a less obtrusive warning in the status bar would also be ok ;-)

GilesBathgate, 2017-12-12:
@dartrax Yes having the warning appear when you press export to G-Code or Save STL makes more sense. However I've made a PR which just displays the error in the status bar, which I think is better. Try it out and see what you think.

lordofhyphens, 2017-12-30:
Closing because #4220 is merged. 

@GilesBathgate FYI if you reference an issue in the commit message like "Fixes #xxxx" Github will close the associated issue when the commit is merged.

closed 2017-12-30 by lordofhyphens

dartrax, 2018-01-22:
Hello,
thanks for your PR, @GilesBathgate. I've tested with 19ea7a5 last week and I've never seen any Warning. This is pretty fine for me but may not be what you intended. In Status Bar I get "Processing triangulated mesh..." and other preview-calculating messages, but any outside-print-bed-warning may get overwritten there.