slic3r/Slic3r #4217: Mouse wheel zoom does not work in 3D-Tab after Drag&Drop-Operation

dartrax, 2017-12-05:
### Version
1.3.0 dev 2017-11-07

### Operating system type + version
Win 7 64bit

### Behavior
* _Describe the problem_
After dropping a .stl-File into 3D-Tab, Mouse wheel zoom is broken until switching to Preview-Tab, zoom there, and switch back to 3D-Tab.
To me it seems to be some kind of control focus bug. It is not possible to set the focus to 3D canvas by clicking it. Instead, focus is taken by 3D canvas when hovering it, but this does only work with Preview Tab canvas, not with 3D Tab canvas.

* _Steps needed to reproduce the problem_
- Open slicer
- drop stl into 3D-Tab
- confirm the "Outside print bed?"-Dialog
- Try to scroll -> no zoom. Control Focus is somewhere else.
- switch to Preview-Tab
- Hover 3D Canvas -> The control focus gets taken by 3D Canvas
- switch back to 3D Tab
- Try to scroll -> zoom is working again.

lordofhyphens, 2018-01-06:
Can't reproduce on d8a331f on Windows 10, 64-bit or 32-bit.

dartrax, 2018-01-22:
Hello,
here is how to reproduce on 19ea7a5, testet with Windows 7 64 bit.

### Steps
1. Open Slic3r 19ea7a5
2. drop stl into 3D-Tab
3. Try to scroll -> zoom works ok
4. Hit "Delete All"
5. again drop stl into 3D-Tab
6. Try to scroll -> no zoom. Control Focus is in Print settings combo and you can't get it back by clicking or hovering the 3D Canvas
7. switch to Preview-Tab
8. Hover 3D Canvas -> The control focus gets taken by 3D Canvas
9. switch back to 3D Tab
10. Try to scroll -> zoom is working again.

andyjbm, 2018-01-23:
Ok. so....

I just discovered something significant here.

So I had a go at Variant A and it was broken as described above.

I minimized all the other apps I had open and just had slic3r and windows explorer across two monitors slic3r on the primary and WE on the secondary. Tried Variant A again. I couldn't break it. ( I had closed slic4r and reopened it. See below.)

So what was different? I went back and looked at what apps I had open.

I had chrome open on my gmail full screen behind slic3r which was sized about 3/4 of the full display over the top. And yep, the zoom wheel broke again.

I've tried various combinations and what I've discovered so far:

Once it breaks, you need to close slic3r and reopen for it to not break again. Closing/minimizing chrome is not enough.

Also, interestingly if you execute variant A with chrome minimised and succesfully defeat variant A (ie the zoom still works) then full sizing chrome behind slic3r DOES NOT BREAK IT ANYMORE for future attempts at variant A.  

In compiling this right now I've got this github open in chrome and it also breaks slic3r zoom so it's not website specific.

So here's a sure way to experience variant A above.

1. Open chrome full screen on Monitor 2 (secondary).
2. Open windows explorer 3/4 screen size over the top of chrome on monitor 2. Navigate to your ,stl
3. Open slic3r on monitor 1 (Primary). It can be full screen size or part screen size, it doesn't matter.
4. Drag your .stl file from windows explorer.
5. Execute Variant A above and zoom will break as described.

Now zoom will ALWAYS break like this no matter what until slic3r is closed and reopened.

I believe this will also occur on a single monitor system if you have chrome open behind slic3r and windows explorer and drag from explorer across chrome and into slic3r. I've not been able to try this yet but once I have some time back home I will and post an update later this week.

Laters,

Andy.

dartrax, 2018-01-23:
This is a very interesting and surprising find...
Infact, that is exactly my setting: Two Monitors, Slicer at one and Chrome at the other, both maximized.

I can nearly confirm your observations with Variant A and Chrome.
Except one thing: 

>  It can be full screen size or part screen size, it doesn't matter.

Slicer must have been maximized once since it started. If you close Slicer not maximized so it will start in restored position, it will not break with Variant A even when Chrome is maximized. But just maximize Slicer once and restore immediately, it will break...

I got curious and wanted to know how does Variant B interfer with Chrome. I came to the result, that Variant B will break when Chrome is running, even when it is minimized - (with the condition that Slicer is maximized, like above).

So next question: Does it even break without chrome at all?
It does! Just execute Variant B twice:

1. Close Chrome
2. Open Slic3r 19ea7a5
3. Be sure to have it maximized
4. drop stl into 3D-Tab
5. Try to scroll -> zoom works ok
6. Hit "Split", you may need to confirm a message window
7. Try to scroll -> zoom works ok
8. Hit "Delete All"
9. again drop stl into 3D-Tab
10. Try to scroll -> no zoom. Control Focus is in Print settings combo and you can't get it back by clicking or hovering the 3D Canvas
11. switch to Preview-Tab
12. Hover 3D Canvas -> The control focus gets taken by 3D Canvas
13. switch back to 3D Tab
14. Try to scroll -> zoom is working again.

lordofhyphens, 2018-01-25:
This is still 99% likely something that isn't fixable due to not being in our codebase.

dartrax, 2018-01-25:
Thanks for reopen. I hope we can find a workaround...

dartrax, 2018-02-14:
I was not able to reproduce this with slicerPE - so it certainly is fixable.