slic3r/Slic3r #4221: Extremely long slicing time (about 30 minutes)

alex-khoroshko, 2017-12-09:
### Version
1.3.0_dev git hash 8ca6171
Checked with 1.2.9 too, problems persists.

### Operating system type + version
Ubuntu 16.04 LTS 

### Behavior
This STL(attached) takes very long to slice - I see one core (latest gen dual core pentium) fully loaded, slicing take about 30 minutes. Other stl's slice fine on my pc, this specific one is problematic. I attached slicing config, but long slicing time took place with any settings I set.

#### STL/Config (.ZIP) where problem occurs
[files.zip](https://github.com/alexrj/Slic3r/files/1545017/files.zip)

bubnikv, 2017-12-09:
What's the time without the support?

lordofhyphens, 2017-12-09:
@bubnikv doesn't look like support has anything to do with it; the delay is coming from the initial mesh processing (at least on my system). 

This model has some neat things going on with it too. I have no idea why performance is so bad.

lordofhyphens, 2017-12-09:
Lots of "Unable to close point having XXX points" messages too.

bubnikv, 2017-12-09:
Maybe it is just the printf of this message.

Anyway, I have rewritten this stage a lot around February this year. I
don't remember the details, but there was some obvious bottleneck. The
Slic3r PE processes this model in around 3 minutes on my laptop.

I think the two important changes are the following, but I don't remember
the details anymore.
https://github.com/prusa3d/Slic3r/commit/1b89c08bfc8c3f930648d2ffdc5a3161f5622144
https://github.com/prusa3d/Slic3r/commit/1909c75c21525c7b7be077456237c392cb54a12a

Also I think these commits fixed some bad data sharing, which made the
parallelization counterproductive in the old code.