slic3r/Slic3r #4222: Slic3r pe constantly freezing on Intel HD graphics

Patrik-svobodik, 2017-12-10:
### Version
1.37.2-prusa3d-win64_

### Operating system type + version
w10 x64

### Behavior
*Constantly freezing when trying to view an stl (sometimes even without stl)
* occurs after start

bubnikv, 2017-12-10:
This is an evergreen, caused by wrong OpenGL driver installed by the Windows updater. The Redmond guys should talk to the guys at Santa Clara more often.

https://github.com/prusa3d/Slic3r/issues/268
https://github.com/prusa3d/Slic3r/issues/233

lordofhyphens, 2017-12-26:
Slic3r PE, not Slic3r. I'll reopen if you reproduce from http://dl.slic3r.org/dev/win and it's caused by anything other than what @bubnikv already pointed out (using wrong OpenGL driver).

closed 2017-12-26 by lordofhyphens