slic3r/Slic3r #4223: custom Start gcode is processed before hotend is at temperature

gec100, 2017-12-10:
### Version
Slic3r-master.2017.12.08.1553.b5276e6.64bit

### Operating system type + version
Win10

### Behavior
Code generated by Slic3r starts with
`M107
M190 S100 ; set bed temperature and wait for it to be reached
M104 S240 ; set temperature`

after that comes the area with custom start Gcode. If that custom Gcode has extrusion in it, like priming the nozzle, can lead to problems cause the nozzle is not at the temperature yet. 

so the code should start with M109 instead of M104. or I'm doing something wrong?

here is what i have in my custom start Gcode:
`G28 ; home all axes
G29 ; auto bed leveling
G1 Z5 F5000 ; lift nozzle
G1 X40 Y2 F3000 ; avoid binder clips
G1 Z0.2 F3000 ; get ready to prime
G92 E0 ; reset extrusion distance
G1 X60 Y22 E10 F600
G1 X80 Y2 E15 F600
G1 X100 Y22 E15 F600
G1 X120 Y2 E15 F600
G1 X150 F5000 ; quick wipe
G92 E0`

and here is the start section of the gcode generated by Slic3r:
`; generated by Slic3r 1.3.0-dev on 2017-12-10 at 19:50:47

; Printer Config Notes: 
; G28 ; home all axes
; G29
; G1 Z5 F3000 ; lift
; G1 X40 Y2 F3000 ; avoid binder clips
; G1 Z0.2 F3000 ; get ready to prime
; G92 E0 ; reset extrusion distance
; ;G1 X120 E10 F600 ; prime nozzle
; G1 X60 Y22 E10 F600
; G1 X80 Y2 E15 F600
; G1 X100 Y22 E15 F600
; G1 X120 Y2 E15 F600
; G1 X150 F5000 ; quick wipe
; G92 E0
; 
; external perimeters extrusion width = 0.40mm (3.02mm^3/s)
; perimeters extrusion width = 0.48mm (7.54mm^3/s)
; infill extrusion width = 0.48mm (10.05mm^3/s)
; solid infill extrusion width = 0.48mm (2.51mm^3/s)
; top infill extrusion width = 0.48mm (1.88mm^3/s)

M107
M190 S100 ; set bed temperature and wait for it to be reached
M104 S240 ; set temperature
G28 ; home all axes
G29 ; auto bed leveling
G1 Z5 F5000 ; lift nozzle
G1 X40 Y2 F3000 ; avoid binder clips
G1 Z0.2 F3000 ; get ready to prime
G92 E0 ; reset extrusion distance
G1 X60 Y22 E10 F600
G1 X80 Y2 E15 F600
G1 X100 Y22 E15 F600
G1 X120 Y2 E15 F600
G1 X150 F5000 ; quick wipe
G92 E0
; Filament gcode
M109 S240 ; set temperature and wait for it to be reached
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0`

#### STL/Config (.ZIP) where problem occurs
[config.zip](https://github.com/alexrj/Slic3r/files/1545963/config.zip)

lordofhyphens, 2017-12-10:
Add `M109 S[first_layer_temperature]; heat nozzle to [first_layer_temperature]C;` to your regular print start gcode.

lordofhyphens, 2017-12-10:
Your observed behavior uncovered a related bug in the software--contrary to what the tooltip said, putting M109/M104 in a filament gcode did not suppress the automatic generation. I've fixed that. You still need to put a `M109` in your printer start gcode if you want it to heat up before doing something with the filament, as the filament gcode is handled after printer gcode.

closed 2017-12-10 by lordofhyphens

gec100, 2017-12-11:
thx @lordofhyphens . but M109, if I understand correctly, is a filament and not printer related code. There is no point to preheat the nozzle to f/e 240 if i'm going to print with PLA. So I'd had to change this custom start gcode every time I change the filament type.
not to mention that auto bed leveling should be done with both nozzle and bed preheated.

lordofhyphens, 2017-12-11:
Using the [] parameter tells it to preheat to (I believe) the active
extruder temperature. So it will use the filament temp profile (one of the
reasons why we didn't have filament custom start code for quite some time).

gec100, 2017-12-11:
ah ... so first_layer_temperature is actually a variable.!? i thought it's just a placeholder for manual value. I'll try that. thx

lordofhyphens, 2017-12-12:
Any config option can be used that way.

commit 576a86d4 (2017-12-10): Consider filament start gcode when deciding whether or not to prepend heating gcode.

Fixes #4223.