slic3r/Slic3r #4225: Top infill layers not connected

Tinchus2009, 2017-12-11:
### Version
1.2.9-1253-g576a86d

### Operating system type + version
win10 64

### Behavior
Top infill layers are not being conected to perimters, usually only the last 2 gets connected, but the previous ones not. This makes bloobs, wich in turn later makes the final top layers have a horrible finish in the borders of the print.
This is confirmed looking at the 3d simulation:
[image]
[falla top.zip](https://github.com/alexrj/Slic3r/files/1547654/falla.top.zip)

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

foreachthing, 2018-01-08:
There's no STL inside your zip file.
Also, try to run your STL through https://service.netfabb.com first.

Tinchus2009, 2018-01-11:
Here it is the stl. It has no errors. Problem can be seen starting at layer 24.68
[Handle.zip](https://github.com/alexrj/Slic3r/files/1621125/Handle.zip)

foreachthing, 2018-01-11:
I can see the same "error" with Slic3r-master.2018.01.08.1579.19ea7a5.64bit !
If you only use 1 perimeter "line", it's better. More then 1 = this funny issue.