slic3r/Slic3r #4226: Retraction settings missing

vovcacik, 2017-12-11:
### Version
1.3.0-dev
Slic3r-master.2017.08.14.1313.a54ca92.64bit

### Operating system type + version
Windows 10 Pro 64 bit

### Behavior
According to the [manual][] there should be *Retraction* section on *Printer* tab. I went through the options and there is nothing about retraction.

[manual]: http://manual.slic3r.org/expert-mode/fighting-ooze

lordofhyphens, 2017-12-12:
It is in the Extruder section.

vovcacik, 2017-12-16:
@lordofhyphens 
[image]

lordofhyphens, 2017-12-17:
That's "Print", not "Printer".

vovcacik, 2017-12-17:
Thanks a lot! This was painful though. You may want to add note about the *Extruder* section to the [manual][]; e.g. *Printer settings* tab > *Extruder 1* page > *Retraction* settings.

[manual]: http://manual.slic3r.org/expert-mode/fighting-ooze

closed 2017-12-17 by vovcacik

lordofhyphens, 2017-12-17:
Manual is in github too (https://github.com/alexrj/Slic3r-Manual) and is in
Markdown format. Make a pull request and it can get fixed faster.