slic3r/Slic3r #4227: Slicer was working ok but now wont open

vwssandcar, 2017-12-12:
### Version
_1.3.0-dev _

_Use `About->About Slic3r` for release versions._

_Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. *Only* use normal Slic3r version IDs._

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### windows 10 64bit
_What OS are you using, and state any version #s_

### Behavior
* _When opening Slic3r it never comes up. loads for 5 seconds then stops trying. When i open the Slic3r Console it closes immediatelyafter it says "caught C++exception of type or derived from 'std: :exception': <unspecified file>(1): '=' character not found in line at C:/Users/alexe/Desktop/3dprinter/Slic3r/lib/Slic3r/config.pm line 99." _
* _try to open program_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _
[image]
_

_Is this a new feature request? No_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

lordofhyphens, 2017-12-12:
>    says "caught C++exception of type or derived from 'std: :exception': (1):
> write-great-feature-request-bug-report/

Go into your user folder and move/delete your Slic3r config directory?

vwssandcar, 2017-12-13:
I could not figure out how to do that so i deleted Slic3r off the computer, again, but this time after I deleted it I searched Slic3r on the computer and deleted every other file related. When I reinstalled Slic3r it worked fine. Just had to reset all my settings. Thanks for the help.

closed 2017-12-13 by vwssandcar