slic3r/Slic3r #4232: Feature request: Fan kickstart

gec100, 2017-12-15:
not sure if this is only my problem but if the starting speed of the fan is below 25 -30% my fan will not start spinning. maybe kick-starting the fan to 100% for a half a second or so and then slowing it down to cooling speed would be good?

VanessaE, 2017-12-26:
Since it's hardware control, this is the domain of your printer's firmware; Marlin has settings for this (referred to as "kickstart", in fact).  What your bot uses likely has as well.

gec100, 2017-12-27:
i've seen this option in simplify3d.

gec100, 2017-12-27:
just checked.. the fan_kickstart_time setting in the Marlin firmware is for controller Fan

VanessaE, 2017-12-27:
No, it's for your layer cooler fan.  I ought to know - I use it. :smile:

VanessaE, 2017-12-27:
No, it's for your layer cooler fan. I ought to know - I use this feature. :smile: ￼

```c
// When first starting the main fan, run it at full speed for the
// given number of milliseconds.  This gets the fan spinning reliably
// before setting a PWM value. (Does not work with software PWM for fan on Sanguinololu)
#define FAN_KICKSTART_TIME 500
```

"Main" fan here refers to your layer cooler fan.  The stuff about "controller fan" found above it in the source code is for a different fan entirely (incidentally, I have my controller fan wired straight to +12v).

gec100, 2017-12-27:
ok. thx. i'll try that.

gec100, 2017-12-30:
it works exactly as you said it would. 
thx

closed 2018-03-06 by lordofhyphens