slic3r/Slic3r #4234: question RE movement

ruggb, 2017-12-16:
### Version
1.3.0-dev
these are the 2 files I have installed - 
slic3r-master.2017.04.03.843.7a4d2bd
Slic3r-master 2017.10.01.a28f431

### Operating system type + version
Win 10-1709
### Behavior
I have been seeking help on this thread.
http://forums.reprap.org/read.php?263,800862,801045#msg-801045
It explains it all. I have concluded it is something Slic3r is doing but I have no clue as to why.

#### STL/Config (.ZIP) where problem occurs
[files.zip](https://github.com/alexrj/Slic3r/files/1564928/files.zip)

lordofhyphens, 2017-12-26:
One thing that looks weird about the model is the multiple shells. If it was properly hollow (one shell and the normal/facings worked out, I don't think Slic3r would think it was multiple objects. As it is, you can hit split and it'll pull out the hollow interior.

That being said, I can't see anything that looks like what you are seeing from the generated toolpath,
[image]

This is with Slic3r-master.2017.11.07.1542.8ca6171.64bit

Here's from gcode.ws of the 10th or so layer:
[image]

lordofhyphens, 2017-12-26:
Fun fact: Netfabb service deletes the inner shell with its repair routine.