slic3r/Slic3r #4235: Wrong/mangled layer height when using the "Z full steps per mm" feature

VanessaE, 2017-12-17:
### Version

commit 202a90ff903be5b61fca317c6b614af0d5f34d8a

### Operating system type + version

Debian 9

### Behavior

Slic3r is not doing the math properly when fitting the layer height in one's chosen "Print" profile to the "whole Z steps per mm" setting in their "Printer" profile.  I have my whole Z steps set to 250 (4000 divided by 16x microstepping), which is 0.0625 mm per whole step with my M5 threaded rod.

But Slic3r is mangling the math somehow, such that I get layers of orderly-but-wrong thicknesses.  For a print with a declared layer height of 0.125 and a first layer of 0.25, in the gcode I get layers at 0.250, 0.256, 0.384, 0.512, 0.640, 0.768, and so on, as though the layer height is being clipped to an even binary value.  As you can see, only the first layer here is correct.  I've omitted Z-lift value from that list (their deltas are correct also).

The print ought to have had layers at 0.250, 0.375, 0.500, 0.625, 0.750, and so on.

If I disable Z full steps feature (set it to 0), I get exactly the layer heights I expect.

#### STL/Config (.ZIP) where problem occurs

no config needed, use any random model.

VanessaE, 2017-12-17:
Updated the issue with layer heights that are correct for my hardware.   Didn't actually change the outcome though.

lordofhyphens, 2017-12-17:
@alexrj is free to correct my usage of the math here by the way.

lordofhyphens, 2017-12-17:
z_steps_per_mm was being used to set layers the model was being sliced up at and with the math involved did not make much sense to me -- the expression resolved to (4 / zsteps). I have changed it to (1 / zsteps * 4), both in adjust_layer_heights and in the gradation application.

@platsch may also want to take a look at it.

VanessaE, 2017-12-17:
@lordofhyphens fixed this in 8f26cd7619bed295f114ac279d6d63169086b400

lordofhyphens, 2017-12-17:
Also I apparently broke something else with it, so there's that.

lordofhyphens, 2017-12-18:
And broke tests for adaptive-slicing. This should get looked at by @platsch, the z gradation application is what is causing the strange behavior and I don't know why there is a 4x multiplier on the layer height when it is supposed to be *full* z steps per mm.

platsch, 2017-12-18:
Seen it, but didn't look into the code yet. I never liked the way the z full steps feature was implemented...
Will do as soon as I find some time.

eavoc, 2017-12-19:
I have problems too. I've set the Z full steps/mm to 25 in order to use the adaptive slicing. But if I set a 0.2 layer height Slic3r makes each layer of 0.16 mm. 
I'm using Slic3r-master.2017.12.15.1558.54205b7.64bit on Windows 10 64 bit.

platsch, 2017-12-20:
OK. First thing: the explanation for the factor 4 is hidden in the original issue #1827 where @nophead argues: 

> For best accuracy you want to move in steps of four times the full step.
> That is when the phase pattern gets back to the same state so if, for
> example, one coil is stronger than the other the position will still be
> consistent.

I always found this to be very opaque to the user and think we should at least mention it in the tooltip or maybe better don't use the factor but mention that a good value for this feature would be 4 * fullstep.

Given this, @VanessaE's setup results in 250 steps/mm -> 0.004 mm/step * 4 = 0.016mm minimum_z_distance. You are trying to achieve 0.1mm, the closest multiples of 0.016 are 0.096 and 0.112. The algorithm chooses 0.096 which is closer to 0.1 ("round down").

In other words: with your configuration it is not possible to achieve exact 0.1mm layers in 4*fullstep intervals.

Thad said, I don't like the way it is. I vote for removing the factor 4 entirely. We can leave a note in the tooltip text and / or in the manual. This is potentially important for people with a belt driven z-axis.

platsch, 2017-12-21:
Created a PR to fix this.
@lordofhyphens with e0a82eee3de44eb83dcd8bdf729b29457ddeac6a you tried to also apply min_dz to first_layer_height. Thats not necessary, the `adjust_layer_height` is just used to "prime" the layer distribution to avoid accumulating errors. The final z-gradation is applied later in `generate_object_layers` and includes the first layer.
The PR also adds some comments and minor changes to clarify this.
I also added a short hint to the possible use of the factor 4 to the tooltip.

closed 2017-12-21 by lordofhyphens

3DMakerArgie, 2020-11-08:
I suspect that I'm having the problem described here. My RepRap M5 rods Z_Axis behave as it should with Marlin 2.0 and Arduino2560+Ramps1.5, but when trying to print a simple 5mm cube pyramid, doesn't rise enough Z. I'll try with slic3r config. Hope it helps.

3DMakerArgie, 2020-11-08:
> Updated the issue with layer heights that are correct for my hardware. Didn't actually change the outcome though.

@VanessaE ¿Would you be so kind to tell me how you solved the issue?¿Which hardware configuration did you correct?

VanessaE, 2020-11-08:
Since the outcome didn't change, I simply stopped using the feature.

3DMakerArgie, 2020-11-09:
Finally found my problem. Was not Slic3r. My problem was in Marlin config Z_MAX_Feedrate wich was 5, and the right value for my hardware is 2.5. This value gave me the right height in the printed piece. I don't have the technical reason that would be very useful to learn a little more, but worked.
Thanks for sharing your experience.

commit 8f26cd76 (2017-12-17): Caught another use of z steps per mm that needs to be clear in its application.

Fixes #4235
PR #4241 (2017-12-21): Removed factor 4 from layer height gradation + comments + minor stuff
Fixes #4235
commit 6dd40091 (2017-12-21): Removed factor 4 from layer height gradation, don't apply min/max layer height limitation after gradation. Fixes #4235 (#4241)