slic3r/Slic3r #4236: Part of symmetric model is missing

vovcacik, 2017-12-17:
### Version

1.3.0-dev
Slic3r-master.2017.08.14.1313.a54ca92.64bit

### Operating system type + version
Windows 10 Pro 64 bit

### Behavior

- part of symmetric model is incorrectly rendered
- missing part is in red
[image]
- it is also completely missing in the solution
[image]
- it works correctly in OpenSCAD and Windows STL viewer
- I can reproduce this problem for higher towers too

#### STL/Config (.ZIP) where problem occurs

[rail.zip](https://github.com/alexrj/Slic3r/files/1566024/rail.zip)

lordofhyphens, 2017-12-17:
Run the model through netfabb service and confirm that Slic3r isn't trying
to repair it.

lordofhyphens, 2017-12-18:
[image]
Netfabb indicates that the surface is not closed, and a pass through the online service fixes it.

3d model preview of the bad file:
[image]

Fixed model:
[image]

[image]

closed 2017-12-18 by lordofhyphens