slic3r/Slic3r #4237: Axes Extend for each item movement

Careknight, 2017-12-18:
### Version
Slic3r-master.2017.12.15.1558.54205b7.32bit

_Use `About->About Slic3r` for release versions._

_Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. *Only* use normal Slic3r version IDs._

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 7 Professional 

### Behavior
Screenshots attached. 
Moving an item on screen causes the axis colours to extend. 
With my configuration and attached stl file added, (I have added the same file twice and then duplicated each of the files to allow me to have two rotated, two not.)
[Defect Screenshots.zip](https://github.com/alexrj/Slic3r/files/1566362/Defect.Screenshots.zip)

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### STL/Config (.ZIP) where problem occurs
[PLA Config and stl.zip](https://github.com/alexrj/Slic3r/files/1566358/PLA.Config.and.stl.zip)

lordofhyphens, 2017-12-30:
FYI you can drag and drop images and they'll display, no need for zip archive for those.

lordofhyphens, 2017-12-30:
I don't really see this as "ugly" or as a UI problem myself, but I suspect I know what's going on here.

[image]

I think the bounding box for all of the objects is what is used for the axis lengths. These values change as you move an object around, so the lines extend and shrink.

I don't see this as a problem because when you are moving objects around, it's reasonable to want to be able to see the axis extend out so that even if you put the objects in weird places, you still have an idea where it is re: the x/y/z axes. 

I'd probably accept a PR if someone could describe why it is a usability problem.

Careknight, 2017-12-30:
From what I remember, the increase in axis length seems related to the movement of an item. My perspective is the axis should remain the same as the selected build dimensions (200x200x200 in my case) as this indicates to the user if the stl item dropped is larger or exceeds the printing capability. This is mainly for the z axis as the graphic for the xy bed space is visible on screen.

The axis is extending when items are being moved within the build space which does not match the use case case you have identified.
A nice implementation of this would be to only extend the axis as far as the item has placed

It is a ui bug, is there a definition of how the axis is supposed to behave?

lordofhyphens, 2017-12-30:
There probably isn't a clear description of how it was intended to behave. The print bed itself in the GUI is there as a guide.

In my playing around with 2 models I saw that the axis settled on a length the size of the distance between two models if the spacing was bigger than the bed.

For me, anyway,  so long as the axis is long enough to be able to tell which side of it a model lies, it is doing its job.