slic3r/Slic3r #4238: DLP Projector - cannot adjust settings or initiate print

bgrig, 2017-12-18:
### Version
Slic3r-1.3.0-dev-576a86d-x86_64 2017-12-10
Slic3r-1.3.0-dev-b5276e6-x64 2017-12-08

### Operating system type + version
Windows 7 Home Premium
Windows 8.1 

### Behavior
The DLP printing functionality is very buggy.

What works:
- Connection to printer (using RAMBo/Marlin) and DLP (DLP4500 by Wintech Digital).
- Can move motors in "Manual Control".
- Everything seems to work in the "Projection" panel.
- The parameters in "Print" panel are adjustable and the estimated print time is updated automatically.
- Can project layers by changing the slider position.
- Can blank the projection using "Stop/Black" button.
- "Export SVG" button.
- "Close" button.

Functions that don't work (doesn't crash when adjusting):
- Changing the "Layer Height" parameters in the "DLP Projector" window does not change thickness of the projected layers. The default values are the values from "Print Settings - Layers and perimeters" window from the main "Plater" window. These values need to be adjusted in that window which updates the values in the "DLP Projector" window, which then updates the total number of layers and the estimated print time in "Projector for DLP" window.
- Adjusting the raft options in "DLP Projector" window does not generate raft layers even though I adjusted "Raft layers" in "Print Settings - Support material" from the main "Plater" window as well.
- Support generation, while it is generated in "Preview" in the main "Plater" window (after adjusting in "Print Settings - Support material", does not get generated in "DLP Projector" window.

Crashes when:
- Similar to the layer height info, the infill information also needs to be set from "Print Settings - Infill" window from the main "Plater" window. Any infill option other than 100% "Rectilinear" results in Slic3r crashing even on basic models such as a cube. Crash message: "3D Printer Slicer application has stopped working".
- Pressing "Print". Crash message: "3D Printer Slicer application has stopped working".

Very interested in helping with testing of any updates of the DLP functionality.

lordofhyphens, 2018-02-25:
The DLP config window currently does nothing. You need, at present, to set config options via profile.

lordofhyphens, 2018-02-25:
The configuration set in SLAPrintOptions.pm looks like it's being discarded and the plater config is re-read in Projector.pm.

lordofhyphens, 2018-02-25:
@bgrig  @n4rki As of 245aac1172ddf0bf0603d03d9e33b5e1e1a050f0 it should propagate the settings from SLAPrintOptions to the projector.

lordofhyphens, 2018-02-25:
Support should work now too; please give it a try.

closed 2018-02-25 by lordofhyphens

lordofhyphens, 2018-02-25:
Infill less than 100% still doesn't quite work (doesn't generate anything for interior paths)

lordofhyphens, 2018-04-01:
#4319 is fixed now, which covers less than 100% infill, by the way.

lordofhyphens, 2018-04-01:
@bgrig 
I retested some things today:

1) Layer height is now honored. 
2) Rafts should function now.
3) Support material is generated.

bgrig, 2018-04-02:
I can confirm that layer height input is saved, rafts are generated, support material is generated, infill is generated!

I am having an issue with projection with the latest builds. The model is scaled down and partially projected at the bottom left of the screen even though the model is auto-placed in the center and the center of my bed is defined as x/2, y/2. Zoom is set to 100%, with 0 offset in x and y. I retested an old build I have 'Slic3r-1.3.0-dev-576a86d-x86_64 2017-12-10'. With this previous build, the projection is not scaled and gets projected properly at the center of the screen.

[image]

lordofhyphens, 2018-04-02:
I'm going to open a separate issue for your retest/regression and lock this issue.

lordofhyphens, 2018-04-02:
#4355