slic3r/Slic3r #4239: first layer speed too fast

merlino1975, 2017-12-20:
### Version
_Version of Slic3r used goes here_ 
the latest
_Use `About->About Slic3r` for release versions._

_Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. *Only* use normal Slic3r version IDs._

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
win 10/linux
### Behavior
* _Describe the problem_
 I have a problem with the speed of the first layer. It is too high
I managed to solve it partially change the value in the "first layer velocity" field in the slic3r settings. then the filling is fine
Instead the problem is when for the characteristics of the object to be printed I have to use the basic reinforcement option. The first layer of the object (the third one of the whole work) goes too fast causing an irregular filling. Thank you.

* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

lordofhyphens, 2017-12-20:
Going to need a screenshot and config export (ctrl-e) for this. Also post the file name for slic3r that you downloaded. "The latest" is useless to anyone as a reference point.

The first layer speed setting should be ironclad.

merlino1975, 2017-12-20:
sorry, thank you, you're right, the truth is that I'm at the far end of my computer. this evening, as soon as I get home, I'll give you all the data you ask me.
In the meantime let me ask you what you mean by the phrase The first layer speed setting should be ironclad. that you can not change? but is it normal that of the 3 layers that the printer performs to make a solid fill layer only the first layer goes too fast causing a bad surface print? the next two layers are slower than the first layer and allow a correct filling.
could it possibly also be a firmware problem?
soon the information requested by you will arrive
thank you for your time

Data: Wed, 20 Dec 2017 05:41:11 -0800

lordofhyphens, 2017-12-21:
There is a specific speed setting for just the first layer.

You can define it as both a % or as a fixed speed.

Slic3r should be using that value (in the gcode output there is a G1 Fxxxx
command that is setting the speed, note them before G1 X Y E moves and
divide by 60 to get the speed in mm/s).

The only reasons slic3r should be going "too fast" at the first layer is
misconfiguration or a bug.

This is why I need a ctrl+e export of the config you were seeing the
problem on (also usually helpful is a copy of a model you are seeing the
problem on); that config helps me to determine if there is corruption or a
misconfiguration and to try it myself so I can look for a bug if everything
looks alright in your configuration. This is also where having a model as
well helps a lot for reproduction purposes.

I have my own first layer speed set to around 30mm/s and I do not see it
faster than that in my own setup.

lordofhyphens, 2017-12-21:
FWIW I am checking for bug reports while I wait for code to compile at my full time job (which is not related to Slic3r at all). 

Limited time on my part is the other reason why I ask people for that much and in that format--if a problem is easy for me to reproduce myself, I am much more likely to work on it.

merlino1975, 2017-12-21:
io immagino che sia difficile conciliare lavoro e passione. In ogni caso, voi avete, e state facendo, un magnifi lavoro.
Ieri ti ho inviato tutto quello che potevo, spero che ti sia arrivato tutto.
Comunque volevo dirti che quel campo che ti hai specificato è già stato modificato da 30 a 15 (la percentuale di velocità di cui tu dicevi) l'ho già modificata perchè a 30 andava troppo veloce allora abbassandola a 15 veniva bene... anche con 25 va troppo veloce...
il problema però che non c'è distinzione tra prima layer del codice e prima layer dell'oggetto.
cioè, se io sotto l'oggetto metto delle layer di rinforzo, per esempio 2 layer, la terza layer sarà la vera prima layer dell'oggetto. Ma questa terza layer non sarà più considerata come prima layer quindi va alla velocità normale... purtroppo la velocità normale della prima layer dell'oggetto è troppo alta rispetto alla seconda e terza layer dell'oggetto... se riesci a vedere il video che ti ho inviato dovresti capire bene. Se il video non ti è arrivato perchè troppo grande, posso inviartelo via wetransfer

Data: Wed, 20 Dec 2017 20:48:41 -0800

merlino1975, 2017-12-21:
yesterday you did not get the mail because I attached you too large documents
so I'll postpone the configuration. I also have a video where the defect is clearly visible. Would you give me an email address where to send it to wetransfert? please...well... the version is version 1.36.2-prusa3d-win64...
and now i send you a the config.ini, the screenshot.. ... ask me for more details...

  
  

# generated by Slic3r 1.36.2-prusa3d-win64 on Wed Dec 20 23:20:54 2017
avoid_crossing_perimeters = 0
bed_shape = 0x0,200x0,200x200,0x200
bed_temperature = 65
before_layer_gcode = 
bottom_solid_layers = 3
bridge_acceleration = 0
bridge_fan_speed = 100
bridge_flow_ratio = 1.1
bridge_speed = 50
brim_width = 0
clip_multipart_objects = 0
complete_objects = 0
cooling = 1
default_acceleration = 0
deretract_speed = 0
disable_fan_first_layers = 3
dont_support_bridges = 1
duplicate_distance = 6
elefant_foot_compensation = 0
end_filament_gcode = "; Filament-specific end gcode \n;END gcode for filament\n"
end_gcode = M104 S0 ; turn off temperature\nG28 X0  ; home X axis\nM84     ; disable motors\n
ensure_vertical_shell_thickness = 0
external_fill_pattern = rectilinear
external_perimeter_extrusion_width = 0
external_perimeter_speed = 35
external_perimeters_first = 0
extra_perimeters = 1
extruder_clearance_height = 20
extruder_clearance_radius = 20
extruder_colour = ""
extruder_offset = 0x0
extrusion_axis = E
extrusion_multiplier = 1
extrusion_width = 0.3
fan_always_on = 0
fan_below_layer_time = 60
filament_colour = #FFFFFF
filament_cost = 0
filament_density = 0
filament_diameter = 1.75
filament_max_volumetric_speed = 0
filament_notes = ""
filament_settings_id = 
filament_soluble = 0
filament_type = PLA
fill_angle = 45
fill_density = 30%
fill_pattern = rectilinear
first_layer_acceleration = 0
first_layer_bed_temperature = 70
first_layer_extrusion_width = 0.6
first_layer_height = 0.3
first_layer_speed = 25
first_layer_temperature = 218
gap_fill_speed = 20
gcode_comments = 1
gcode_flavor = reprap
infill_acceleration = 0
infill_every_layers = 1
infill_extruder = 1
infill_extrusion_width = 0.6
infill_first = 0
infill_only_where_needed = 0
infill_overlap = 15%
infill_speed = 35
interface_shells = 0
layer_gcode = 
layer_height = 0.2
max_fan_speed = 100
max_layer_height = 0
max_print_speed = 80
max_volumetric_extrusion_rate_slope_negative = 0
max_volumetric_extrusion_rate_slope_positive = 0
max_volumetric_speed = 0
min_fan_speed = 35
min_layer_height = 0.07
min_print_speed = 10
min_skirt_length = 0
notes = 
nozzle_diameter = 0.3
octoprint_apikey = 
octoprint_host = 
only_retract_when_crossing_perimeters = 1
ooze_prevention = 0
output_filename_format = [input_filename_base].gcode
overhangs = 1
perimeter_acceleration = 0
perimeter_extruder = 1
perimeter_extrusion_width = 0.3
perimeter_speed = 35
perimeters = 3
post_process = 
print_settings_id = 
printer_notes = 
printer_settings_id = 
raft_layers = 0
resolution = 0
retract_before_travel = 2
retract_before_wipe = 0%
retract_layer_change = 0
retract_length = 2
retract_length_toolchange = 10
retract_lift = 0
retract_lift_above = 0
retract_lift_below = 0
retract_restart_extra = 0
retract_restart_extra_toolchange = 0
retract_speed = 40
seam_position = aligned
serial_port = 
serial_speed = 250000
single_extruder_multi_material = 0
skirt_distance = 9
skirt_height = 1
skirts = 2
slowdown_below_layer_time = 5
small_perimeter_speed = 15
solid_infill_below_area = 70
solid_infill_every_layers = 0
solid_infill_extruder = 1
solid_infill_extrusion_width = 0.6
solid_infill_speed = 35
spiral_vase = 0
standby_temperature_delta = -5
start_filament_gcode = "; Filament gcode\n"
start_gcode = G28 ; home all axes\nG1 Z5 F5000 ; lift nozzle\n
support_material = 0
support_material_angle = 0
support_material_buildplate_only = 0
support_material_contact_distance = 0.2
support_material_enforce_layers = 100
support_material_extruder = 1
support_material_extrusion_width = 0.6
support_material_interface_contact_loops = 0
support_material_interface_extruder = 1
support_material_interface_layers = 3
support_material_interface_spacing = 1
support_material_interface_speed = 100%
support_material_pattern = pillars
support_material_spacing = 2.5
support_material_speed = 35
support_material_synchronize_layers = 0
support_material_threshold = 50
support_material_with_sheath = 1
support_material_xy_spacing = 50%
temperature = 213
thin_walls = 1
threads = 2
toolchange_gcode = 
top_infill_extrusion_width = 0.45
top_solid_infill_speed = 35
top_solid_layers = 3
travel_speed = 100
use_firmware_retraction = 0
use_relative_e_distances = 0
use_volumetric_e = 0
variable_layer_height = 1
wipe = 0
wipe_tower = 0
wipe_tower_per_color_wipe = 15
wipe_tower_width = 60
wipe_tower_x = 180
wipe_tower_y = 140
xy_size_compensation = 0
z_offset = 0

lordofhyphens, 2017-12-21:
You can drag/drop ZIP files on github interface. Also pictures. Not video, but that isn't particularly useful right now with a copy of the configuration.

One last thing: please use a download from https://dl.slic3r.org/dev (top of list is newest) and see if it still a problem.

From what I can tell with your configuration it should not be going faster than 25mm/s on the first layer, 35mm/s max (support speed), and 100mm/s travel  (non printing) speeds.

You can confirm this by opening the .gcode file and looking at the different F#### statements, particularly before the G1 Z0.5 statement. Divide by 60 to translate to mm/s.

Setting the verbose gcode option in Print Settings -> Advanced will help you understand what the gcode means in context.

lordofhyphens, 2017-12-21:
I prefer all data related to an issue report be deposited here on GitHub.

merlino1975, 2017-12-21:
I'll watch. The problem is that that value (25) I have already changed several times, in the end the value 15 is the one that gives me better results ... (the value 25 I had inserted it to do some tests)

but I would not focus on that value. I'm interested in the value of the first layer of the object. Because that's what I think is not correct. By changing the velocity value before the layer that manages to decrease the speed of the first layer does not mean having solved the problem.
I try to explain it to you with a question question: we consider that to create a full and compact fill sli3cer orders the printer to make 3 layers, the speed of these 3 layers must be the same? if the answer is yes, then there is something wrong with the first layer because it is much faster than the other 2 layers

probably being Italian, I can hardly explain myself well in English, so maybe at this moment I can not make you understand what exactly it is that in my opinion does not work well. In addition to English, I do not understand the programming language either. Ask me to read the gcode for me it could be very difficult, but I will try to do it too.

p.s .: I will try to use the latest version of sli3er even if I do not know how to insert it into a repetier-host, software that gave me the geeetech, brand of the 3d printer.

thanks thanks thanks 

Data: Thu, 21 Dec 2017 02:10:35 -0800

  
  

p.p1 {margin: 0.0px 0.0px 0.0px 0.0px; font: 12.0px Helvetica; min-height: 14.0px}
p.p2 {margin: 0.0px 0.0px 0.0px 0.0px; font: 12.0px Helvetica}

lordofhyphens, 2017-12-21:
Okay, I've taken your config and a recent build of the Slic3r that I support. It looks like verbose gcode was already on (this is good). 

For this snippet, I've taken just the gcode pertaining to the first layer that have a speed parameter in them. I'll go through each one in turn. 

* `G1 Z0.300 F6000.000 ; move to next layer (0)`
    * This is a travel move (no printing occurs). Your config says travel at 100mm/s. 100*60 = 6000.
* `G1 E-2.00000 F2400.00000 ; retract`
    * This is an extruder retraction move. This does not affect the XY speed (because the speed is reset before the printer moves again). Your config says retract at 40mm/s. 40*60 = 2400.
* 'G1 X83.325 Y82.459 F6000.000 ; move to first skirt point'
    * This is a travel move (no printing occurs). Your config says travel at 100mm/s. 100*60 = 6000.
* `G1 E2.00000 F2400.00000 ; unretract`
    * This is an extruder retraction move. This does not affect the XY speed (because the speed is reset before the printer moves again). Your config says retract at 40mm/s. 40*60 = 2400.
* `G1 F1500`
    * This is followed by printing moves (commands that tell the printer to move X, Y, and E). Your defined first layer speed in the configuration is 25mm/s. 25*60 = 1500. 
* `G1 X83.849 Y82.713 F6000.000 ; move to first skirt point`
    * Another travel move.
* `G1 F1500`
    * Print speed set (after travel move)
* `G1 E18.88211 F2400.00000 ; retract`
    * Retraction
* `G1 X108.629 Y108.629 F6000.000 ; move to first perimeter point`
    * Another travel move.
* `G1 E2.00000 F2400.00000 ; unretract`
    * Retraction
* `G1 F1500`
    * Print speed set (after travel move and retraction)
* `G1 X109.164 Y109.164 F6000.000 ; move to first perimeter point`
    * Another travel move.
* `G1 F1500`
    * Print speed set (after travel move)
* `G1 X109.700 Y109.700 F6000.000 ; move to first perimeter point`
    * Another travel move.
* `G1 F1500`
    * Print speed set (after travel move)
* `G1 X109.410 Y109.622 F6000.000 ; move inwards before travel`
    * Another travel move.
* `G1 E14.68451 F2400.00000 ; retract`
    * Retraction
* `G1 X108.036 Y108.150 F6000.000 ; move to first infill point`
    * Another travel move.
* `G1 E2.00000 F2400.00000 ; unretract`
    * Retraction
* `G1 F1500`
    * Print speed set (after retract and travel move)

With the model I provided and a supported version of Slic3r, the gcode output is correct. I cannot find evidence of a bug in Slic3r.

Reproduction:
Using slic3r 1.3.0-dev as of d27b8a933e8cc318d7bbdd8acc7d9d8aef943ce6. 
Gcode export, Slic3r config, simple 20mm box STL.
[4239.zip](https://github.com/alexrj/Slic3r/files/1579765/4239.zip)

You can perform this analysis with whatever model you saw the problem on yourself. 
1. Export gcode from Slic3r. 
1. Open the output gcode file in a text editor.
1. Find the comment `move to next layer (0)`
1. Copy all of the text from there until the comment `move to next layer (1)`
1. Check every line that has a `F` (capitalized) in it. 
    * Divide the number by 60 to get the speed in mm/s. 
    * Compare to relevant parameter in Slic3r config.
    * If there isn't a comment on a line, look at the comments on the lines above and below. 
    * `skirt` means it is expecting to print a skirt, `infill` infill, `perimeter` one of the perimeter types.
1. Every printing move should occur at a speed of `1500` (25mm/s).

Finally:
> Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. Only use normal Slic3r version IDs.

Closing. I will reopen if you can provide a zip archive of the model that causes the declared speed settings in Slic3r to be not what the configuration says they should be and confirm that it is happening with one of the downloads from https://dl.slic3r.org.

If you can only get it to happen with the Prusa3d edition of Slic3r (this is what your configuration says it is from), you should got to http://github.com/prusa3d/slic3r/issues and report it there.

closed 2017-12-21 by lordofhyphens