slic3r/Slic3r #4240: [Feature Request] Show and/or Export layer

gege2b, 2017-12-20:
Hi there

It could be useful to have the possibility to Show / Export only the current layer's Gcode/SVG/PNG/Whatever
This could be done by right-click on the layer cursor in the preview window, witch would show a menu

The main usage for this feature is for peoples that wish to export 2D slices for, i.e.  infographics, 2D plotter, some kind of art.
I agree that Slic3r is a 3D slicing software and this kind of feature isn't really useful in this scope

cheers

lordofhyphens, 2017-12-21:
This would probably be 90% ui work, for any enterprising people. Slic3r
already will export a SVG of the slice.

gege2b, 2017-12-21:
Well, I can try to add it but the last time I messed with the UI, it took me a few hours to figure out how to add a button and two checkboxes xD

SVG Slice is good but miss the infill pattern, at least for what I plan to do ( #4213 )

lordofhyphens, 2017-12-21:
Yeah I have not had time to figure out why sparse infill is crashing SVG slicing.