slic3r/Slic3r #4242: 3d view pan often moves view wrongly

VanessaE, 2017-12-26:
### Version

commit 6dd400914ee08fe122c5e28d6304d95d5cb2cffc

### Operating system type + version

Debian 9

### Behavior

Through some combination of rotate, zoom, and pan while examining parts on the plate, eventually Slic3r gets things all screwed up, and starts moving the view wrongly in response to mouse pan actions.  That is, I might try to right-click and drag the view right and the visuals might move UP instead.

Needless to say, if I drag right, then under no circumstances should the visuals do anything except move right to follow my mouse.

I can't figure out how to reliably "reset" Slic3r's behavior when it gets like this, short of quitting and restarting it.

Left-click to rotate/pitch the view seems to work fine in any case.

#### STL/Config (.ZIP) where problem occurs
config:  [config-20171226.ini.zip](https://github.com/alexrj/Slic3r/files/1587835/config-20171226.ini.zip)
Use any random model, though this can probably be reproduced without anything loaded.

lordofhyphens, 2017-12-30:
Hrm, I'm having trouble reproducing this on Windows :(

VanessaE, 2017-12-30:
Yeah, it's tricky to hit the right combo of movements; I'm still not sure what the trick is, all I know is that somehow I'm always triggering it.