slic3r/Slic3r #4243: Slic3r crash with thin infill sections.

craftydan, 2017-12-27:
### Version
1.3.0-Dev 8ca6171

### Operating system type + version
Win7 32, SP1

### Behavior
May be related to issue #4221.

Loaded the attached STL, and swapped over to preview (by accident) at 0% infill.  Took longer than expected, but did complete.  I saw the single walls on the upper tubes, so I increased the fill . . . 

Filling this STL with any pattern tends to trigger the offending behavior.  When switching to preview with any pattern selected and above 0% infill:

- The toolbar will stall at "processing triangulated mesh . . ."
- the process becomes laggy in updating the GUI (OS may throw a "Not responding" if an event it monitors is fired) 
- The process will consume all 4 cores to 100% (with the OS breaking in for monitoring, naturally). 

Eventually, it will generate "An unhandled exception occurred" dialog with abort-retry-ignore options (no other details).  Clicking "ignore" will populate the preview platter with a short model (missing a significant number of top layers).  

The pattern selected seems to have an effect relationship with the number of layers missing (where the exception is sensed) with the more complex patterns throwing the exception at lower layers, though I'm not certain the time it takes is any different.  

Following the ignore, the GUI is very laggy and any re-slicing exhibits unstable behavior (unhandled exception).  When closing the GUI the OS will generate a visual c++ runtime library error (terminated in an "unusual way"), and an OS level "application has stopped working" dialog with the below listed in the details:  

Problem signature:
  Problem Event Name:	APPCRASH
  Application Name:	Slic3r.exe
  Application Version:	1.3.0.0
  Application Timestamp:	5a013b69
  Fault Module Name:	libstdc++-6.dll
  Fault Module Version:	0.0.0.0
  Fault Module Timestamp:	3488346f
  Exception Code:	40000015
  Exception Offset:	00020c48
  OS Version:	6.1.7601.2.1.0.768.3
  Locale ID:	1033
  Additional Information 1:	9d39
  Additional Information 2:	9d3904fbb0e4b7201c74f88fd69a4d67
  Additional Information 3:	2592
  Additional Information 4:	2592a0d79b13eb1e85536b2b72e3f1c4

I've re-tested this with the latest win32 dev build (6dd4009) with practically the same outcome.

#### STL/Config (.ZIP) where problem occurs

[Twisted_concentric_pencil_stand.zip](https://github.com/alexrj/Slic3r/files/1588211/Twisted_concentric_pencil_stand.zip)

[config.txt](https://github.com/alexrj/Slic3r/files/1588216/config.txt)

lordofhyphens, 2017-12-27:
Yeah, the app crash info is next to useless :/

As a quick sanity check, have you tossed the model through service.netfabb.com?

craftydan, 2017-12-27:
I expected as much, but someday somebody will find it handy . . . in the mean time, it's easier to ignore useless information than use data that was left out of the report.

I had not, but it's a fair check.  No dice.  The "fixed" STL exhibits the same behaviors.