slic3r/Slic3r #4245: Skirt extrusion width issue

xsrf, 2017-12-29:
### Version
1.2.9

### Behavior
As far as I can tell, the Skirt is always extruded with the nozzle with. This leads to an unnecessary thin extrusion on small nozzles (I use a 0.25mm nozzle but usually extrude at 0.5mm) which sometimes doesn't stick well because filament flow is also not the best at the beginning and thus the skirt is sometimes dragged inwards, messing up the print later.
It's not really an error, but for sure an issue sometimes.

### Solution
It would be nice if skirt extrusion width could be set separately or is at least set to the first layer extrusion width.

lordofhyphens, 2017-12-29:
Odd, should be either support width or first layer.

xsrf, 2017-12-29:
You can't even set the support extrusion width when support is disabled, so I don't think it is used here.
In my case I have set default extrusion width to 0.5 and all other extrusion widths to 0. This generates a much thinner skirt with nozzle size set to 0.25 than with nozzle size set to 0.4.

lordofhyphens, 2017-12-29:
Is this a problem with 1.3.0-dev? Https://dl.slic3r.org/dev

lordofhyphens, 2017-12-30:
In 1.3.0-dev the skirt width is set to the first layer extrusion width if it's not 0 (automatic), the perimeter extrusion width otherwise. 

Here's the code snippet that governs it (Print.cpp:789):
````
    ConfigOptionFloatOrPercent width = this->config.first_layer_extrusion_width;
    if (width.value == 0) width = this->regions.front()->config.perimeter_extrusion_width;
````

FWIW the same logic is used for brims. 

When printing regions the logic has as fall-through case to the default extrusion width, but this isn't replicated for the skirt and brim code.

If you set a first layer perimeter width it would probably work (which is why this behavior hadn't been noticed, most people have a first layer width configured).

lordofhyphens, 2017-12-30:
@xsrf there, it should default to the default extrusion width for the first configured object in both cases. This also affects the brim.

commit a1a2a31e (2017-12-30): Use the default extrusion width for the first loaded object if the configured perimeter width and first layer width is 0.

Fixes #4245
closed 2017-12-30 by lordofhyphens