slic3r/Slic3r #4246: Cooling and Speed reduction for external Line

arcosh, 2017-12-29:
Hi,
I am using Slic3r 1.3.0-dev and have a principal question/issue. 

When I am using in Filament Setting the cooling function. And defined that under a certain time per layer my printing speed should be reduced. Should this not also happened for the external Layer?
This is not happening. In my gcode the inner two lines are reduced in speed and the external Line is defined by Print Settings Speed.
Attached a example gcode...
Best Regards

M106 S255
G1 Z32.750 F7800.000
G1 X11.373 Y-7.467 F7800.000
G1 F3579.25 <- reduced in speed
G1 X11.373 Y7.467 E1582.43000
G1 X11.030 Y7.415 E1582.44919
G1 X10.236 Y7.212 E1582.49455
G1 X9.612 Y6.986 E1582.53133
G1 X8.873 Y6.632 E1582.57668
G1 X8.300 Y6.285 E1582.61374
G1 X7.761 Y5.889 E1582.65080
G1 X7.151 Y5.341 E1582.69616
G1 X6.703 Y4.851 E1582.73294
G1 X6.212 Y4.195 E1582.77829
G1 X5.868 Y3.627 E1582.81507
G1 X5.514 Y2.888 E1582.86043
G1 X5.288 Y2.264 E1582.89721
G1 X5.085 Y1.470 E1582.94257
G1 X4.985 Y0.813 E1582.97935
G1 X4.941 Y-0.000 E1583.02442
G1 X4.985 Y-0.813 E1583.06950
G1 X5.085 Y-1.470 E1583.10628
G1 X5.288 Y-2.264 E1583.15164
G1 X5.514 Y-2.888 E1583.18842
G1 X5.868 Y-3.627 E1583.23377
G1 X6.212 Y-4.195 E1583.27055
G1 X6.703 Y-4.851 E1583.31591
G1 X7.155 Y-5.345 E1583.35297
G1 X7.649 Y-5.797 E1583.39003
G1 X8.305 Y-6.288 E1583.43539
G1 X8.873 Y-6.632 E1583.47217
G1 X9.612 Y-6.986 E1583.51752
G1 X10.236 Y-7.212 E1583.55430
G1 X11.030 Y-7.415 E1583.59966
G1 X11.314 Y-7.458 E1583.61552
G1 X11.875 Y-8.029 F7800.000
G1 F3579.25 <- reduced in speed
G1 X11.875 Y8.029 E1584.50451
G1 X11.635 Y8.016 E1584.51782
G1 X10.930 Y7.908 E1584.55733
G1 X10.088 Y7.693 E1584.60542
G1 X9.417 Y7.450 E1584.64494
G1 X8.634 Y7.075 E1584.69303
G1 X8.021 Y6.704 E1584.73270
G1 X7.443 Y6.279 E1584.77237
G1 X6.797 Y5.699 E1584.82046
G1 X6.315 Y5.172 E1584.85997
G1 X5.795 Y4.477 E1584.90806
G1 X5.425 Y3.866 E1584.94758
G1 X5.050 Y3.083 E1584.99567
G1 X4.807 Y2.412 E1585.03518
G1 X4.592 Y1.570 E1585.08327
G1 X4.484 Y0.865 E1585.12279
G1 X4.438 Y-0.000 E1585.17072
G1 X4.484 Y-0.865 E1585.21866
G1 X4.592 Y-1.570 E1585.25817
G1 X4.807 Y-2.412 E1585.30626
G1 X5.050 Y-3.083 E1585.34578
G1 X5.425 Y-3.866 E1585.39387
G1 X5.795 Y-4.477 E1585.43338
G1 X6.315 Y-5.172 E1585.48147
G1 X6.799 Y-5.701 E1585.52114
G1 X7.328 Y-6.185 E1585.56081
G1 X8.023 Y-6.705 E1585.60890
G1 X8.634 Y-7.075 E1585.64842
G1 X9.417 Y-7.450 E1585.69651
G1 X10.088 Y-7.693 E1585.73602
G1 X10.930 Y-7.908 E1585.78411
G1 X11.635 Y-8.016 E1585.82363
G1 X11.816 Y-8.025 E1585.83362
G1 X12.300 Y-8.476 F7800.000
G1 F3600 <- non reduced in speed
G1 X12.300 Y8.476 E1586.48033
G1 X11.592 Y8.439 E1586.50739
G1 X10.845 Y8.324 E1586.53620
G1 X9.963 Y8.099 E1586.57093
G1 X9.253 Y7.842 E1586.59975
G1 X8.432 Y7.449 E1586.63448
G1 X7.785 Y7.057 E1586.66333
G1 X7.175 Y6.609 E1586.69218
G1 X6.498 Y6.001 E1586.72691
G1 X5.988 Y5.443 E1586.75573
G1 X5.443 Y4.715 E1586.79045
G1 X5.051 Y4.068 E1586.81927
G1 X4.658 Y3.247 E1586.85400
G1 X4.401 Y2.537 E1586.88282
G1 X4.176 Y1.655 E1586.91755
G1 X4.061 Y0.908 E1586.94636
G1 X4.013 Y-0.000 E1586.98106
G1 X4.061 Y-0.908 E1587.01575
G1 X4.176 Y-1.655 E1587.04457
G1 X4.401 Y-2.537 E1587.07930
G1 X4.658 Y-3.247 E1587.10811
G1 X5.051 Y-4.068 E1587.14284
G1 X5.443 Y-4.715 E1587.17166
G1 X5.988 Y-5.443 E1587.20639
G1 X6.499 Y-6.001 E1587.23524
G1 X7.057 Y-6.512 E1587.26409
G1 X7.785 Y-7.057 E1587.29882
G1 X8.432 Y-7.449 E1587.32764
G1 X9.253 Y-7.842 E1587.36236
G1 X9.963 Y-8.099 E1587.39118
G1 X10.845 Y-8.324 E1587.42591
G1 X11.592 Y-8.439 E1587.45473
G1 X12.240 Y-8.473 E1587.47949
G1 X12.112 Y-8.123 F7800.000
G1 Z33.000 F7800.000
G1 X11.373 Y-7.341 F7800.000
G1 F3489.74 <- reduced in speed

lordofhyphens, 2017-12-29:
Please attach a small test model and a copy of the exported configuration you are using. Make a ZIP archive of the two and drop into the reply box on GitHub. 

It may also be helpful for you to turn on verbose gcode in Print Settings -> Output Options

arcosh, 2017-12-29:
here the Files... 
[Files.zip](https://github.com/alexrj/Slic3r/files/1593904/Files.zip)

Area5142, 2018-01-02:
I think this issue is related to _Bug? slowing affects not equally to all speeds #4185_.

I thought this was a bug too, but it is intentionally done to keep perimeter speed the same and make a consistent surface finish.
I would suggest that slow down affected the hole layer (as before) and there was an option to turn it off for external perimeters.
The fast external perimeters gives me a lot of problems when printing TPU and flex with small layers.

kyrreaa, 2020-01-28:
I am experiencing this very thing right now and it is ruining my print.
There should be an option to allow everything to be reduced without having to reconfigure job speeds for perimiter. What happens to my little model now is the internals is slowed down a lot to allow cooling (estimated 15mm/s) , but then the outer perimiter is slapped on at 100mm/s and the result is massive issues with extrusion pressure and cooling time not maintained. The model in question is a small dia pulley so the outer perimiter length is very long vs the internal volume. This increases the effect.