slic3r/Slic3r #4247: [feature request] randomized speeds on short segments to reduce resonance.

sfKyale, 2017-12-31:
Randomly* changing speed for short segments between an upper and lower bounds could reduce the likely hood of resonance. 

*The set of speeds should have a high standard of distribution without repetition for the duration of a print. 

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

lordofhyphens, 2017-12-31:
This is something you should probably test with just gcode modifications in some situations to see if it actually would help.

Given the sparseness of your description and everything else that needs doing, someone needs to do the experimental work to see if it is beneficial at all, define tradeoffs, etc.

sfKyale, 2017-12-31:
Thank you. You are correct.