slic3r/Slic3r #4248: Tiny pieces of infill between perimeters, generating around holes is incontinuous

Amunak, 2017-12-31:
### Version
1.3.0-dev (2017-12-21-master 64-bit)

similar behavior also present in 1.2.9

### Operating system type + version
Windows 10

### Behavior

Slic3r seems to be pretty bad at generating holes and related features. I'd expect that if a hole's perimeter is connected to some other perimeter, those would be printed *together* so that issues with first layer adhesion of holes are minimized.

Basically if I have a pad with 4 holes close to its perimeter I'd expect Slicer to try to merge both perimeters in such a way as to make it as continuous as possible. Instead all perimeters are printed separately, so even when they touch or even overlap one perimeter is printed first, then the holes' perimeters, etc. making adhesion harder.

There are also some tiny, weird infill areas created inside those perimeters. Again, I'd expect Slicer to either completely omit those and just add a little more extrusion to "fill the gaps" or I'd expect it to do its best to make the lines continuous or whatever, but I don't understand why it generates lines of infill that have like 0.5mm in length and 0.1mm in width.

Essentially I'd really appreciate if Slicer generally tried to print continuous features of same type actually continuously. The perimeters may be generated by distinct features, but if they touch they are basically one feature, right? It would not only make adhesion way easier, but also reduce print time because there would be less wasted travel for no reason.

Or am I perhaps overlooking something?

### Examples

[image]

[image]

[image]

#### STL/Config (.ZIP) where problem occurs

Example test files: [test-files.zip (18KB)](https://github.com/alexrj/Slic3r/files/1595290/test-files.zip)

My config for completeness: [Slic3r_config_bundle.zip (1.8KB)](https://github.com/alexrj/Slic3r/files/1595292/Slic3r_config_bundle.zip)

lordofhyphens, 2017-12-31:
It's a pretty hard problem in the general case. If you do not want to see
the gap fill then turn off gapfill.

Amunak, 2018-01-01:
Sure, I'd say all of slicing is fairly hard and complicated.

In this case the simplest optimization would probably be to find all overlapping perimeters and instead of printing them in sequence one after another finding an optimal path to print  them "together" so that at least the adhesion problem goes away.

Tinchus2009, 2018-01-01:
Would this be a work aroud (dont know if it would work): What about useing the post processing script published in ht epostprocessing folder in the dev tree, the one called decimate.pl. Thta post scriips is suposed to removed tiny lines, it would work?