slic3r/Slic3r #4251: Save everything (models, plates, configurations - also unsaved) in one (factory) file

hoegge, 2018-01-04:
### Feature request
It would be great if you could save the plate and all active settings in a single "factory" file, so you can transfer everything and save for future retrieval later even if any configurations has been changed. This should also include settings that are not saved in a print, filament or printer configuration.

### Explanation / details
I have just "discovered" Slic3r after having used Simplify3D for a while. Slic3r is in many ways much more intuitive especially how the settings are organised. One thing I really miss is the ability to save a "factory" file that contains everything for a specific print. It saves all settings, STLs and plate arrangements, so you know you can accurately do the exact same print again at some time later without having to spend a lot of time checking what are the right settings. Especially when you have had to tweak and change settings specifically for a model to make it just right, then you don't want to save a separate config for every print to be able to reproduce it and then a "factory file" would be very useful.

bubnikv, 2018-01-05:
Slic3r could export the platter into an AMF file, and it could export the
working configs into a single config file.  The two files together (AMF and
config) make up your factory file.
Ideally Slic3r would have an option (or do it by default) to save the
config into the AMF file.

The Prusa fork could also import the config from a G-code file.

hoegge, 2018-01-05:
@bubnikv Do you by "could" mean "can"? I finally understood that the "config bundle" are all the presets that you export or import and "config" is the current settings. So `"config + plate = factory file"` ? In that case I assume that the feature request is not necessary and and sorry for the inconvenience.

hoegge, 2018-01-05:
But the Layers table in the settings for each object and the settings per part, they are not saved anywhere are they? So you lose those from print to print?

lordofhyphens, 2018-01-05:
The per part settings should be saved in AMF already (I think 3MF too but I
can't remember).

hoegge, 2018-01-05:
But the adaptive layer height and the layer height tables cannot be / are not saved, right?

hoegge, 2018-01-05:
Can see it is kind of a duplicate of #3545 - so maybe should just be tracked there. Would be nice though.

platsch, 2018-01-09:
At least the adaptive layer height table is currently not saved to AMF files, I'm not sure about the old region based table. This is a known (important) issue and on my todo-list...

lordofhyphens, 2018-01-09:
I don't think the old table is saved either.