slic3r/Slic3r #4252: Building Slic3r on Windows for Distribution

Torbenuz, 2018-01-05:
Windows 10 64bit. 
Strawberry perl 5.26 32bit 
boost 1.63

I have modified Slic3r commit: 3a29aeccd92f545a1fbfbf73cde1df00ae9afd10
and followed the instructions:

https://github.com/alexrj/Slic3r/wiki/Building-Slic3r-on-OSX-and-Windows-for-Distribution

compiled it and i can start the gui and use ist. Now I want to pack it.

I used `pp -M Encode::Locale -M Moo -M Thread::Semaphore -M OpenGL -M Slic3r::XS -M Unicode::Normalize -M Wx -M Class::Accessor -M Wx::DND -M Wx::Grid -M Wx::Print -M Wx::Html -M Wx::GLCanvas -M Math::Trig -M threads -M threads::shared -M Thread::Queue -l C:\strawberry\perl\site\lib\auto\Wx\Wx.xs.dll -o -l C:\strawberry\perl\site\lib\Alien\wxWidgets\msw_3_0_2_uni_gcc_3_4\lib\wxbase30u_gcc_custom.dll -l C:\strawberry\perl\site\lib\Alien\wxWidgets\msw_3_0_2_uni_gcc_3_4\lib\wxmsw30u_core_gcc_custom.dll -l C:\strawberry\perl\site\lib\Alien\wxWidgets\msw_3_0_2_uni_gcc_3_4\lib\wxmsw30u_gl_gcc_custom.dll -l C:\strawberry\perl\site\lib\Alien\wxWidgets\msw_3_0_2_uni_gcc_3_4\lib\wxmsw30u_adv_gcc_custom.dll -l C:\strawberry\perl\site\lib\Alien\wxWidgets\msw_3_0_2_uni_gcc_3_4\lib\wxmsw30u_html_gcc_custom.dll -o slic3r.exe slic3r.pl`

I had to change the paths for Slic3r::XS, Wx, Wx::DND, Wx::Grid, Wx::Print, Wx::Html to static one cause it would not be found. 

It start produces the exe. But the exe instantly closes after Launch.

I have some warnings, but i am not sure how to resolve them and if I even have to:
`# Use of runtime loader module Module::Runtime detected.  Results of static scanning may be incomplete.
# Use of runtime loader module Module::Implementation detected.  Results of static scanning may be incomplete.
Use of uninitialized value in pattern match (m//) at C:/Strawberry/perl/site/lib/PAR/Packer.pm line 796.
Use of uninitialized value in pattern match (m//) at C:/Strawberry/perl/site/lib/PAR/Packer.pm line 796.
Use of uninitialized value in pattern match (m//) at C:/Strawberry/perl/site/lib/PAR/Packer.pm line 796.
Use of uninitialized value in pattern match (m//) at C:/Strawberry/perl/site/lib/PAR/Packer.pm line 796.
Use of uninitialized value in pattern match (m//) at C:/Strawberry/perl/site/lib/PAR/Packer.pm line 796.
Use of uninitialized value in pattern match (m//) at C:/Strawberry/perl/site/lib/PAR/Packer.pm line 796.
Use of uninitialized value in pattern match (m//) at C:/Strawberry/perl/site/lib/PAR/Packer.pm line 796.`

Hope anyone can tell me where to look next.

lordofhyphens, 2018-01-05:
Read through appveyor.yml (reference appveyor.com docs) and the referenced files in the package dir, it is those commands that the build server uses to produce the packaged builds at dl.slic3r.org/dev/win

lordofhyphens, 2018-01-05:
If your work is useful to more than just you, consider a pull request for the changes.

lordofhyphens, 2018-07-09:
No response, so assuming they sorted themselves out.

closed 2018-07-09 by lordofhyphens