slic3r/Slic3r #4253: File from Thingiverse makes Slic3r 1.3.0-dev crash all the time

ilikegreen, 2018-01-06:
### Version
_Current 1.3.0-dev_

### Operating system type + version
_Linux (based on current Ubuntu LTS, also tried the Windows version)_

### Behavior
* _https://www.thingiverse.com/thing:1806407 the bottom files make Slic3r crash_

#### STL/Config (.ZIP) where problem occurs
_https://www.thingiverse.com/thing:1806407_

lordofhyphens, 2018-01-06:
Give it a pass through service.netfabb.com first.

Any model on thingiverse is garbage until proven otherwise.

ilikegreen, 2018-01-06:
But a program shouldn't just die with an error like
`*** Error in `./perl-local': realloc(): invalid next size: 0x00000000064c5fc0 ***`
because of a bad input. This is a potential security flaw.
It wouldn't be a problem if just a message occurred that this file coulnd't been opened or if the rendering or slicing wouldn't work, but that's something different.

The file works with outdated Slic3r versions and other slicer software btw.

lordofhyphens, 2018-01-07:
Admesh is the underlying library used for input since 1.2.x. Replacing it has been on my wish list for quite some time, but when someone is working on this program in their spare time, crashes caused by garbage input (and such have a quick workaround) are low on the priority list.

Also STL is a terrible file format.

lordofhyphens, 2018-01-07:
I will happily accept a pull request with a fix.

lordofhyphens, 2018-01-17:
By the way, cova119_bottom_logo.stl didn't crash on my dev system.