slic3r/Slic3r #4254: Z-lift.

rewolff, 2018-01-08:
### Version
1.3.0-dev
1.2.5-2-g28d7b0d

### Operating system type + version
Linux, ubuntu 16.04

### Behavior
The PROBLEM is that my delta printer fails to print properly: The head scrapes the print during moves. This means that the "stick to the bed properly" has to be even better than normally, and that it causes my printer to lose steps every now and then. 

Not knowing it is called "Z LIFT" I could not find a setting for this. So I wrote a program to take any sequence "go from A to B" moves together and convert that into one move-up, move, move down. 

Looking at the resulting G-code I found that the "avoid borders" was still active, I think that's unnecessary. But I found the gui element to disable that. 

But slic3r will issue a "move up vertically", do the move, "move down vertically". For the motion planner, those right angle corners are difficult and slow. My code allowed the X and Y moves to start by giving the move a target at 1mm along the route from A to B at the higher Z-height. This allows the printer to take the corner at the speed where it can "suddenly start or stop" the Z movement. Even better would be to fit a smooth curve (a parabola would work) that has a certain steepness when leaving A and coming in at B. In my code I used a for now constant ratio: 2 (horizontal) : 1 (vertical). 

Now it still failed the print. :-(, so I am now trying with Slic3r's built-in Z-lift instead of mine. Maybe the 2:1 was still too steep. 

#### STL/Config (.ZIP) where problem occurs
```
G1 X36.393 Y-17.242 Z7.000 F7200
G1 X8.975 Y34.748 Z7.000 F7200
G1 X8.509 Y35.633 Z6.500 F7200
```

lordofhyphens, 2018-01-08:
So your printer is not properly calibrated?

https://ladvien.com/robots/kossel-mini-calibration/

Your parabolic lift doesn't make much sense for a cartesian.

lordofhyphens, 2018-01-08:
There is also firmware retraction available that will move in Z.

rewolff, 2018-01-08:
You're right. The "smooth path" is only for delta machines that can do the Z movement almost for free. 

Yes, I've been through the calbration procedure several times. But somehow, my hardware tends to catch on the object when traveling at the same height. 

Similarly, I read somewhere today that this LIFT would not help get working prints. Well, when you print an overhang in PETG or ABS, that overhang is going to curl up. This can be as much as a few mm. But when you simply follow the outline of the object, depositing more plastic on top continues to work. This causes the overhang to become more and more stiff and results in a reasonable print after some time. 

Possibly you'll get a little bit less plastic than according to the design file, but other than that a working print. But not if the head bumps into the curling plastic and bumps the object off the platform or looses a few steps in the process. 

I started the print with a 1mm Z-lift this morning and the "go up, move, go down" was very visible. Just moving already at in the right direction at a 45 degree angle is very likely to work and is just a little bit faster.

lordofhyphens, 2018-01-08:
One other thing that is machine specific that you may wish to look into is
extrusion tuning. It is easy to over extrude enough to catch the head like
you say.

For your feature request, I will consider a pull request to make it an
option.

rewolff, 2018-01-08:
Re: Pull request: Yeah, I noticed. I considered asking for what that  "Pull Request or Bust" meant until I looked it up for myself. 

You're right. I might be overextruding a bit. On the other hand, I've calbrated that already and I thought I had that figured out. With KLIPPER firmware, there is simply a configuration file and not the whole "either save it to the eeprom or edit the source of the firmware" mess.

rewolff, 2018-01-08:
Update: From a quick scan of the code, the code is not structured in a way to make this easy: The lift is called from "retract" at which point the destination of the move is not available. I'll see what happens next. 
* I am wrong, and the current move-destination is still easily available.
* I can restructure the code. 
* I can give up. :-)

lordofhyphens, 2018-01-08:
AFAIK the next destination is not available currently to the retract code.

Making it available is an exercise to the reader ;)

You should start by making a new copy of retract and use it instead. Less likely to break things, and it can be refactored later to support both usages.

lordofhyphens, 2018-01-08:
Specifically nobody has needed to know until now. 

The usual flow at things like layer change is to lift, move, unlift. Parabolic sounds neat but I would be cautious about collisions.

rewolff, 2018-01-08:
I think the code is structured wrong. I think there should be a "do a (travel) move from A to B" function. 
That does: 
* retract if configured/necesssary
* lift if configured/necessary
* move
* unlift
* unretract. (tract? :-) ) 

So, somehow you seem to think that the conditions for "lift" are always exactly the same as for retract. Depending on the cost for each, wouldn't it be useful to be able to specify the conditions separately?

The current way, by calling "lift" from the retract and unretract means that you can't lift if you're not retracting. As I you must know, a small "lift" is as good as free on a delta printer. You can cut the "cost" of doing lifts in two by doing the lift across the first 1-2 mm of travel, and maybe the "straight down" in the current implementation isn't that bad: you eliminate the risk of still hitting a border during the unlift.

lordofhyphens, 2018-01-08:
The Venn diagram for retract and lift has a large overlap between the two. Whether or not lift is a subset of retract I don't know ;).

You retract to relieve pressure. Even on a delta not retracting during a lift will ooze filament. 

Your proposed travel move consists of:
GCodeWriter::retract()
GCodeWriter::lift()
GCodeWriter::travel_to_xy()
GCodeWriter::unlift()
GCodeWriter::unretract()

https://github.com/alexrj/Slic3r/blob/master/xs/src/libslic3r/GCodeWriter.cpp

rewolff, 2018-01-09:
> Whether or not lift is a subset of retract I don't know ;).
Agreed. But even IF you and I agree that they should always happen together, that doesn't mean that someone else doesn't think it is useful to do it differently! 

With mosfet packages like "DPAK2" (it's the surface mount version of TO220 if that rings a bell) the thermal conductivity towards the PCB is good and towards the exposed side of tha package it is worthless. So worthless that manufacturers do NOT specify the value. They have measured it and it is so worthless that it doesn't need a specification. Well that has left me with "well if they don't tell me, I'll have to measure it myself". But before that happened I found a manufacturer who DID tell in their datasheet. 

Similarly, if WE decide it is worthless to allow people to lift-but-not-retract that doesn't mean that people should be forced to follow that rule. When would that strategy be worthwhile? Maybe when lift is cheap and can be done without any penalty on speed. And maybe retract is not necessary when someone has a low-drip nozzle with high retract-penalty. 

Anyway, last night's print failed again. With the bottom half looking perfect, the nozzle ended up with two pieces of the third quarter of my print hanging off it and "filament mess" for the last quarter. 

I may still be having overextrusion. I don't know where it is coming from: I calibrated the mm extrusion per step quite accurately. But the failed print measured 25g while Slic3r predicted 16.6 . 

When I changed the extrusion multiplier from 0.92 to 0.86 the gcode was not recalculated and when I sent it to the printer, I still had 16.6 predicted grams of extrusion. Then while I was trying to force a recalculate (after cancelling the print) slic3r crashed. The new instance calculated 15.x worth of filament.

rewolff, 2018-01-09:
```
GCodeWriter::retract()
GCodeWriter::lift()
GCodeWriter::travel_to_xy()
GCodeWriter::unlift()
GCodeWriter::unretract()
```
Right! And when it's written like that, the "travel_to_xy ()"  has to know what XY. So now its easy to make the "lift" aware of the direction in which to initiate a move.