slic3r/Slic3r #4255: Adaptive Slicing changes overall thickness of parts

foreachthing, 2018-01-08:
### Version
Slic3r-master.2017.12.18.1561.d27b8a9.64bit

### Operating system type + version
Windows 7 x64

## Setting 1
* Got some parts, all 3 mm in thickness.
* Sliced it using the _adaptive slicing_ option - quality set to **80%**.

### Result
* Sliced parts are now **2.96 mm** thick.

## Setting 2
* Got a part, 3 mm in thickness.
* Sliced it using the _adaptive slicing_ option - quality set to **81%**.

### Result
* Sliced parts are now **3.04 mm** thick.

# Expected Behavior
* Same thickness for all quality settings!

Unless the quality setting is set to 0 or 100% (sorry, didn't try everything between 0 and 100.0), the thickness is _wrong_.

lordofhyphens, 2018-01-08:
@platsch

lordofhyphens, 2018-01-08:
@foreachthing please provide the usual: a set of exported configurations (ctrl-e for a single config, not a bundle), and a 3d model you see the issue on.

platsch, 2018-01-08:
Could be a spline-related problem...
Generally: for static slices, the thickness of an object can't be matched exactly unless it happens to be a multiple of the layer thickness. The adaptive algorithm should provide better, but in some cases still not exact results. Please provide the model for verification.

foreachthing, 2018-01-09:
@lordofhyphens Sorry for my laziness not uploading any model nor config file. This was with all models so I didn't bother.
So here is "any" model with the config file. Different model, same issue.
[cube10x10x3.zip](https://github.com/alexrj/Slic3r/files/1614215/cube10x10x3.zip)

As @platsch mentioned, this must be a math problem. 
Oh, and set the quality to zero (with this model) I only get 2.9 mm thickness this time.
Please let me know if you need more info.

platsch, 2018-01-09:
Thx, should do. Will check this the next days.

lordofhyphens, 2018-01-09:
Thank you. Even "all models do this" should include a simple model ;)

platsch, 2018-02-26:
This is a general problem related to the discretization into 2D-layers.
As I said, for static layers, the last layer generally doesn't match the top of an object, unless the user sets a layer height which does.

In your example, the adaptive algorithm tries to reduce the layer thickness towards the top to resemble the original model as precise as possible, but doesn't care about the top.

You can use the `Match horizontal surfaces` setting which also considers the object's top as horizontal feature and tries to shrink / widen the last layers for a better match. Unfortunately, this has the side effect that Slic3r will try to match every horizontal surface in the model which is the reason why the feature is disabled by default.

I just opened a PR #4321 to fix this at least for some cases. With that change, the adaptive layer generator will always try to exactly fit the last layer.

For your example, even that won't help for all configurations. Example: for a high quality the layer thickness will be the minimum printable height for the last few layers. If the last layer happens to end slightly above the top, it is not possible to furter reduce it without violating the `min_layer_height` constraint. So we would have to slightly increase the last few layers to be able to remove the last layer which is not easy to do and would also affect the print-quality in an undesired manner.

I suggest we merge the PR and accept that discretization errors will always occur. Improving specific situations will probably have too many side-effects on other configurations.

lordofhyphens, 2018-02-26:
@foreachthing can you give the PR from @platsch a try and see it helps out?

foreachthing, 2018-02-27:
Well, it fits my models really nicely (±0.04 mm) and better than before.
Before it was like ±0.1 mm. Min. 0.1 and max. 0.3 mm layer thickness. Haven't tried much else yet ...

lordofhyphens, 2018-05-09:
@foreachthing It's merged into 1.3.1-dev.

lordofhyphens, 2018-05-09:
Let me know if I can close this issue.

closed 2020-07-16 by foreachthing