slic3r/Slic3r #4256: [Feature request] Combine initial Z and XY moves

nmaggioni, 2018-01-09:
I moved back to Slic3r after having used Cura for some time, and noticed a slightly different behavior when starting a print: with Cura the initial Z and XY moves are combined in a single command, whereas in Slic3r they are separate moves.

Given that I lower my bed (or lift the nozzle, the concept is printer-independent) before starting a print to better inspect it, this leaves nasty oozing traces on my printbed while the nozzle moves from home to the initial printing position.

Wouldn't it be handy if this was done in a single, smooth move? _This FR is similar to #964._

---

### Example:

+ Go from this:

```gcode
M109 S200 ; set temperature and wait for it to be reached
[...]
G1 Z0.200 F7200.000 ; move to next layer (0)
G1 E-2.00000 F2400.00000 ; retract
G1 X115.327 Y80.449 F7200.000 ; move to first skirt point
G1 E2.00000 F2400.00000 ; unretract
G1 F3000
G1 X116.806 Y79.659 E0.05678 ; skirt
```

+ To this:

```gcode
M109 S200 ; set temperature and wait for it to be reached
[...]
G1 E-2.00000 F2400.00000 ; retract
G1 X115.327 Y80.449 Z0.200 F7200.000 ; move to next layer (0) and to first skirt point
G1 E2.00000 F2400.00000 ; unretract
G1 F3000
G1 X116.806 Y79.659 E0.05678 ; skirt
```

BloodBlight, 2018-02-05:
I second this!  Not being able to do this just ends up making a mess on the print bed (and preventing the first layer from sticking) if you have a more "ozzy" setup.  :<

nmaggioni, 2018-02-06:
@foreachthing Thanks for the help, but I'm using Linux and I'd rather avoid calling in Mono on this to run an external C# postprocessor.

I'll come up with a little Python script in the meantime, but in an ideal future I'd like to avoid postprocessing at all :)

BloodBlight, 2018-02-06:
@nmaggioni Same, was debating it as well, if you do, would you share it?

nmaggioni, 2018-02-07:
@BloodBlight Absolutely! I quickly hacked something together: https://github.com/nmaggioni/slic3r-smooth-first-move - it has not been field tested yet but the output file looks good.

At the moment it works for skirts only as that's what I use most of the time; extending it to support other structures (nothing, rafts, etc) should be easy.

foreachthing, 2018-02-07:
> ... but in an ideal future I'd like to avoid postprocessing at all ...

And somebody else want's it different than we do. 

OT: .... I hate Python: every software that uses python comes with its own Python distro. So, today, I have 35 python.exe on my c-drive alone.

lordofhyphens, 2018-02-07:
@foreachthing sounds like a Windows problem :D

McPcholkin, 2019-04-22:
I also have issues after moving from cura to slic3r, i not find ready solution so i made my own.
I move first Z move to first X and Y move, so it make first move on all directions (and not scrape curved bed...).
It written on python so it must work on any platform, tested only on windows. 
https://github.com/McPcholkin/Slic3rPE