slic3r/Slic3r #4257: Whining Sound When Layer 2 Starts

tbedore, 2018-01-11:
### Version
1.38.7

### Operating system type + version
macOS High Sierra - 10.3.2

### Behavior
* When I would start printing, everything would work fine, and then layer 2 it would start printing a little, and then it would stop making a loud whining sound and would just be stuck on the layer, not doing anything, the read-out on the LCD was normal.  
* I sliced something in Slic3r and in order to see the preview, you need to hit slice now.  So you can slice anything in the newest drivers, hit slice now, and then export the g-code.  That is all I did, and it failed on the 2nd layer, when I went to go check it, the printer was running at 235 and it was cooling down from a 250 temp.  
* The printer should print and continue the whole print, and then finish it.  
* It fails on a part of layer 2, ONLY IN SLIC3R
 

#### STL/Config (.ZIP) where problem occurs
[Archive.zip](https://github.com/alexrj/Slic3r/files/1624150/Archive.zip)

lordofhyphens, 2018-01-11:
You're using Prusa3d fork of Slic3r. That's not supported here.

Reproduce your problem with 1.3.0-dev (https://dl.slic3r.org/dev/mac/slic3r-1.3.0-dev-19ea7a5.dmg) and I'll reopen.

lordofhyphens, 2018-01-11:
As for your issues, make sure your printer limits your max speed for Z if it's a prusa-style printer. Slic3r expects you to have limits in the firmware that are reasonable for speeds so it doesn't have to have X/Y/Z speeds in its configuration (travel speed is travel speed in all 3 dimensions). 

The regular version of Slic3r (see above) reslices when you switch back to preview after making changes.

closed 2018-01-11 by lordofhyphens

tbedore, 2018-01-11:
whoops!  I thought since I came here from the prusa github that this would be their slic3r oh well.

lordofhyphens, 2018-01-11:
Obviously if you can reproduce with ours then I would like to hear it.