slic3r/Slic3r #4258: 2018 Feature Request - Outlines height / infill height separate , Beautifull object for the same time of low quality printing

studiodyne, 2018-01-12:
Hi , this is just for inspiration, just an idea but if Slic3r Devs make it , it's a big new release for 2018 that will up the slic3r to the first position of this category of software

Imagine , Slic3r have separate settings for outlines and infill , and the intelligence to combine these
For example  0.1mm for outlines and 0.4 for infills
The object will have an amazing looking and the time to make it is approximatively the same of a low quality printing. All overhangs and many angles are processed in high resolution too

The process need to be intelligent
Two slicing process
One with the high layer height  and the ' REST not printed ' will be sliced with the lower height
The slicer combine all , and the print can begin

Advantage of a high resolution for all curved roof or overhangs and more , and the high resolution is just used where is needed , and the rest is printed with high height

Thks

platsch, 2018-01-12:
as already pointed out in the Marlin firmware tracker https://github.com/MarlinFirmware/Marlin/issues/7597#issuecomment-356276369 this is implemented and used by many people for years.

Just set `Combine infill every` to a number > 1.
The idea is not new, the first description I'm aware of can be found in Emmanuel Sabourin master thesis "Adaptive High-Precision Exterior, High-Speed Interior, Layered Manufacturing" http://citeseerx.ist.psu.edu/viewdoc/summary?doi=10.1.1.556.1100

supermerill, 2018-01-15:
duplicate of #1201 ?

platsch, 2018-01-15:
@studiodyne can we close this or do you still see a missing point?

studiodyne, 2018-01-15:
It's ok , and i'm happy , to see , the idea , is already in many brains , i will look at the other posts here

You can close , and many thanks for devs jobs 
Very good job !

closed 2018-01-15 by studiodyne