slic3r/Slic3r #4259: Feature Request - Multiple head overlap/area reduction adjustment per layer

kfriesth, 2018-01-13:
Allow the system to define printable area with overlap/area reduction ie (6.35 mm had better fea analysis) to allow overlap of one head and reduction of area on a second head with a per layer adjustment and then reversed area size on the following layer and then repeated. This will allow the layers to stagger the joint in a brick laying style so each layer never has an aligned joint which removes the issue of joint layer weakness for larger objects using multiple heads.

lordofhyphens, 2018-01-14:
Yeah, I can't make heads or tails of this. 

Please include some pictures to better illustrate what you want.

kfriesth, 2018-01-16:
This method with the bed size known will allow any multiple head/multiple laser to adjust the assigned area for the particular head to vary in a preset area adjustment so that no two layers in a row will have a joint to align, thereby allowing machine size to increase while parsing data for the defined area's to the appropriate head or laser.
[image]