slic3r/Slic3r #4260: Gap in Perimeter of holes or Hole Perimeters not Connected.

wirlybird, 2018-01-13:
Slic3r PE 1.37.2
Window 7 64bit

My apologies if this is not ok to ask about here but thought it is a general slicer behavior question and am just hoping for your thoughts on this issue.  Thank you!

Ran into an odd issue in Slic3r.  See the first picture from Slic3r, the areas within the blue circles, the perimeter of the holes are not connected.
Now, in the second picture from S3D inside the red circles you can see the hole perimeters are connected.
I have tried variations of number of perimeters, extrusion width etc. but no change on the one in Slic3r.
The model has no errors running through a fixer.

Thanks for any thoughts.
Might just be an anomaly in Slic3r.

[image]
[image]

[Kitchen drain plug Non Disposal.zip](https://github.com/alexrj/Slic3r/files/1628219/Kitchen.drain.plug.Non.Disposal.zip)

lordofhyphens, 2018-01-14:
You need to reproduce this behavior on a build from dl.slic3r.org/dev. This is not the Prusa3d fork.

Closing until this happens.

lordofhyphens, 2018-01-14:
Also, turn on thin wall detection if you haven't.

closed 2018-01-14 by lordofhyphens

wirlybird, 2018-01-14:
I found the answer.  Turning off Detect Thin Walls fixed the problem.  Just the opposite that you would expect.
I appreciate the response.

Sorry for a post about the Prusa version but I thought this might be a more general usage issue and could help someone else out if an answer were found.

lordofhyphens, 2018-01-14:
It is a general rule here and written down in the text you deleted to write the issue. 

We don't have the resources to run down issues on the Prusa3d fork, so we ask that all reported issues are reproduced on one of our builds.

Usually it does not take much time at all to do so and we provide binaries for Windows, Linux, and OSX.

Obviously if it is an issue in main Slic3r it belongs here.

wirlybird, 2018-01-14:
Thank you again for the input.  I do appreciate it.
When I get some time I will load one of your builds on a machine and see if I can reproduce what I found and report back to you.
I know you have much more important issues to deal with but I am happy to try to provide hopefully useful and proper feedback.

Thanks again.