slic3r/Slic3r #4261: Add arbitrary G-Code at defined Z heights (in this case for nozzle cooling fan control)

tekikai, 2018-01-15:
As a feature request, I would like to be able to input *arbitrary G-Code at defined Z heights*. The need is there because I have a *second fan to cool the nozzle*. If the print is running, I could enter M106 P0 (P1 is the main fan for the cooling block) S128 for a half speed nozzle cooling fan as an example but I do not want to sit there and watch to enter it when necessary.

This would be good for several things;
- leave the nozzle fan off for a defined number of first layers so adhesion is better
- this goes together with bridge detection (but also is independent of it): either have the nozzle cooling fan on all the time after a few layers or switch it on only when there are overhangs / bridges
- it would also allow to switch other accessories (bed lighting, stepper cooling fan etc.) on and off as needed during the print
- also I am missing an option to control a second fan in the settings at all

It could be integrated with the layer height settings that already exist per object but it would probably be more flexible using its own tab so it would be independent of the layer height settings (or toggleable 'bound to layer height settings: on/off'.

lordofhyphens, 2018-01-15:
Slic3r supports "Conditional Gcode", a preprocessor function that covers your needs. In earlier versions of Slic3r the same functionality could be gotten with some clever use of markers and a post process script.

Fan cooling already is able to be turned off for the first number of layers.

lordofhyphens, 2018-01-15:
https://github.com/alexrj/Slic3r-Manual/blob/master/src/advanced/conditional-gcode.md

https://github.com/alexrj/Slic3r/wiki/Conditional-Gcode-Syntax-Spec

https://github.com/alexrj/Slic3r/issues/3390

lordofhyphens, 2018-01-15:
https://github.com/alexrj/Slic3r-Manual/blob/master/src/expert-mode/cooling.md

tekikai, 2018-01-16:
Hi and thanks for the links, that is pretty much what I am looking for. Re the fan, I did not realize the settings for auto-cooling were for the additional fan, thanks for pointing out that it is already there. I just did not realize it, my mistake!
How would I go about to insert g-code somewhere in the file? As far as I can see, there is 'start', 'end', 'before layer change' and 'after layer change'. And where would I find the names for the slicer settings (other than layer_num or layer_z)?

lordofhyphens, 2018-01-16:
Export a config.ini from Slic3r. That will get you started.

lordofhyphens, 2018-01-16:
Before layer change and after layer change would be your likely candidates.

tekikai, 2018-01-16:
ok, so before and after layer change mean any time before or after and not 'directly before / after'?
Exported it, thanks, that does nicely!

lordofhyphens, 2018-01-16:
Exactly as it says on the tooltip: before layer change inserts that gcode before every layer's print. After puts it after that specific layer.

B0AB1AB2AB3A... where # is a layer's gcode, B is the before layer gcode, and A is the after layer gcode.

closed 2018-01-17 by lordofhyphens