slic3r/Slic3r #4262: Feature Request: Bridge Angle, Infill Density as Functions of Z Dimension

streamin, 2018-01-16:
### Version
1.3.7.2-prusa3d-win64
### Feature Request
1) Allow the user to set different bridge angles for ranges of Z dimension.
2) Allow the user to set different infill density for different ranges of Z dimension.

In the first three frames of the image below Slic3er behaves as expected.  In the fourth frame the bridge angle is deficient.  Allowing the bridge angle to be set for different Z dimension ranges will allow the user to manually fix deficiencies.
[image]

Allowing the infill density to be set for different Z dimension ranges will allow the user to add or subtract strength to their part where necessary.  Depending upon the infill pattern solid infill layers may be necessary when changing densities.

lordofhyphens, 2018-01-16:
Is this a duplicate of #3333?  If not, have you looked at modifier meshes at all?

lordofhyphens, 2018-01-16:
Also, I've changed the "affected" version. Please do not report Prusa3d versions on this tracker (this is not Prusa3d).

This happens to be common between the two, but there's no guarantee that Prusa3d fork will take the code that gets implemented here.

streamin, 2018-01-16:
> Is this a duplicate of #3333? If not, have you looked at modifier meshes at all?

It appears the second half of this feature request is a duplicate of #3333.

> Also, I've changed the "affected" version. Please do not report Prusa3d versions on this tracker (this is not Prusa3d).
>
>This happens to be common between the two, but there's no guarantee that Prusa3d fork will take the code that gets implemented here.

Understood.

lordofhyphens, 2018-01-16:
@streamin did you want manual control, or did you want it to automatically gradient up (which is the core of #3333). Manual control is easily done with modifier meshes (it's one of the original use cases for modifier meshes--changing the infill% for a specific area of a model), either here or in the prusa3d fork.

bubnikv, 2018-01-19:
the prusa fork implements a bridge angle override for some time and it
could be applied to a modifier mesh.

<https://github.com/alexrj/Slic3r/issues/4262#issuecomment-358090234>, or mute
the thread