slic3r/Slic3r #4264: Segfault when "cut"-ing model

pilliq, 2018-01-17:
### Version
1.2.9 (release version)

### Operating system type + version
OSX Yosemite 10.10.5

### Behavior
1. Import model and rotate 90 degrees around y-axis
[image]
[image]

2. Perform cut approximately halfway through the model
[image]

3. Segfault. From OSX crash reporter:
```
Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x00000000fffffff8

VM Regions Near 0xfffffff8:
--> shared memory          00000000ffff4000-00000000ffff5000 [    4K] r-x/r-x SM=SHM  
    

Application Specific Information:
Performing @selector(controlAction:) from sender wxNSButton 0x120e3be0

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   XS.bundle                     	0x0474d776 Slic3r::TriangleMeshSlicer::make_expolygons_simple(std::vector<Slic3r::IntersectionLine, std::allocator<Slic3r::IntersectionLine> >&, std::vector<Slic3r::ExPolygon, std::allocator<Slic3r::ExPolygon> >*) + 710
1   XS.bundle                     	0x0474dc45 Slic3r::TriangleMeshSlicer::cut(float, Slic3r::TriangleMesh*, Slic3r::TriangleMesh*) + 869
2   XS.bundle                     	0x047cbf6d Slic3r::ModelObject::cut(double, Slic3r::Model*) const + 733
3   XS.bundle                     	0x0454f799 XS_Slic3r__Model__Object_cut(interpreter*, cv*) + 841
4   libperl.dylib                 	0x000e74a6 Perl_pp_entersub + 2294
5   libperl.dylib                 	0x000e5af6 Perl_runops_standard + 22
6   libperl.dylib                 	0x00074ca5 Perl_call_sv + 725
7   Wx.bundle                     	0x0596fe67 wxPliEventCallback::Handler(wxEvent&) + 1111
8   libwx_baseu-3.0.0.0.0.dylib   	0x06427d9f wxAppConsoleBase::CallEventHandler(wxEvtHandler*, wxEventFunctor&, wxEvent&) const + 95
9   libwx_baseu-3.0.0.0.0.dylib   	0x0656f7a8 wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) + 104
10  libwx_baseu-3.0.0.0.0.dylib   	0x0656ff98 wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 88
11  libwx_baseu-3.0.0.0.0.dylib   	0x06570904 wxEvtHandler::TryHereOnly(wxEvent&) + 52
12  libwx_baseu-3.0.0.0.0.dylib   	0x065709ac wxEvtHandler::ProcessEventLocally(wxEvent&) + 60
13  libwx_baseu-3.0.0.0.0.dylib   	0x06570a21 wxEvtHandler::ProcessEvent(wxEvent&) + 81
14  libwx_osx_cocoau_core-3.0.0.0.0.dylib	0x0600ddd2 wxWindowBase::TryAfter(wxEvent&) + 114
15  libwx_baseu-3.0.0.0.0.dylib   	0x06570b11 wxEvtHandler::ProcessEvent(wxEvent&) + 321
16  libwx_baseu-3.0.0.0.0.dylib   	0x065703d7 wxEvtHandler::SafelyProcessEvent(wxEvent&) + 39
17  libwx_osx_cocoau_core-3.0.0.0.0.dylib	0x0600cbfb wxWindowBase::HandleWindowEvent(wxEvent&) const + 27
18  libwx_osx_cocoau_core-3.0.0.0.0.dylib	0x05e4c2f8 wxControl::ProcessCommand(wxCommandEvent&) + 24
19  libwx_osx_cocoau_core-3.0.0.0.0.dylib	0x05e173a5 wxButton::OSXHandleClicked(double) + 69
20  libwx_osx_cocoau_core-3.0.0.0.0.dylib	0x05ede6c5 wxOSX_controlAction(NSView*, objc_selector*, objc_object*) + 53
...
```

### Notes
* Model is manifold:
[image]

* Cutting through the center of the model *without* rotating around y-axis succeeds
* Cutting off-center (e.g. 1/4-way) through the model *with* rotating around y-axis succeeds
* I tried cutting the model from #3953 and reproduced a segfault, but the stack trace was different from my model's trace.

#### STL/Config (.ZIP) where problem occurs

[model.zip](https://github.com/alexrj/Slic3r/files/1640060/model.zip)
[config.txt](https://github.com/alexrj/Slic3r/files/1640062/config.txt)

NateTG, 2018-01-17:
Can you try putting into something like meshlab and simplifying the mesh?

lordofhyphens, 2018-01-17:
Or even service.netfabb.com

pilliq, 2018-01-17:
I tried simplifying with meshlab before (quadratic edge collapse decimation) and it still segfaulted.

Just uploaded to netfabb, but got an error while processing: `FAILED: failed to load mesh (#103)`

lordofhyphens, 2018-01-17:
Wow, that'd be an unhappy mesh then. Interesting.

pilliq, 2018-01-21:
Ah no worries. I cut it in half in a different 3D software then imported it into slic3r and it sliced perfectly. It seems more like an edge case for a pretty weird mesh. Thanks for your work on slic3r, it's a great piece of software.