slic3r/Slic3r #4265: Feature Request: half-circle (or arbitrary angle circle) skirt

tekikai, 2018-01-19:
### Version
1.2.9

### Behavior
* What I would love to see as a config option is to be able to print a skirt that draws just half a circle (or any angle). It would need two options: circle segment in degrees, rotation of circle (along z-axis)
* The rationale behind it: I currently print PLA filament which likes to be printed fairly hot (for PLA, 220° first layer, 210° subsequent ones) and I print on a delta printer (Anycubic Kossel Delta). The printable height is about 30cms so after priming the nozzle, it takes quite a while for the nozzle to reach the print bed and start the skirt. By the time it reaches the bed, there is some ooze (even if I remove it by hand a few centimeters before it reaches bed height) that gets carried away when printing the skirt. I usually watch very closely for any stray ooze strings that would mess up my first layer and remove them quickly. The easiest would be if I had a non-circular skirt so that I simply could pull the skirt off the bed which would take any thin ooze strings with it but since it encloses the print object, pulling it off the bed is not possible. If the skirt were half a circle, aiming at the front side of the printer, I could simply pull it off. I am not quite sure yet if the stringing etc. results from diameter problems of the filament etc. (still need some calibration) but in almost every case, if I remove any (even fine) ooze strings while the printer is at the first layer, the print comes out fine. Any strings left behind get tangled around the nozzle and deposited later on which then subsequently messes up the print.

_Is this a new feature request?_
yes

tekikai, 2018-01-24:
Ok, so far I've added a centimeter square 'cube' with the thickness of the first layer (0.3mm) to the print and prepended it to the main object (tried 2cm² too but 1 was enough to clear the nozzle, 2 was overkill).
This made it much easier to catch any stray ooze. So my feature request still stands for a freely choosable skirt object to put on the print bed, maybe even as a default option.

lordofhyphens, 2018-01-24:
See https://github.com/alexrj/Slic3r/blob/master/lib/Slic3r/Print.pm#L227 for the skirt generation routine.

tekikai, 2018-01-24:
Thanks for the link / hint at the skirt generation routine. I wrote a little perl script years ago and the syntax is easy enough to read. If I brush up on it, I might try my hand at implementing it, but a bit unlikely currently (RL constraints etc.).
I see that the request is a bit out of the ordinary (maybe a bit specific to my printer / filament). 
Could you elaborate on the other uses you hinted at? My idea would be not to scrap the skirt printing but to provide the other forms of 'skirt' as an option in the skirt settings.

lordofhyphens, 2018-01-24:
http://manual.slic3r.org/expert-mode/skirt

There's a few uses for a skirt.

Heat retention
### 
This prints a skirt all the way up the print to maintain temps.

Multiextruder purge
###
In multiextruder prints, one of the uses of the skirt is to purge unused toolheads.

tekikai, 2018-01-24:
Ok, I see that for example a multilayer skirt could protect from drafts (ABS printing etc.) and that with multiple extruders it would not be trivial to arrange one or several single-layer cubes as skirt for example, especially. if there are many objects on the plate. A skirt around all objects would be much more convenient there, also because it does not take away so much space in one spot of the build plate.
As mentioned, maybe the arbitrary skirt objects could be added as an option.

Since I got my printer only recently and am pretty new to printing, I will experiment some more. I fear that my filament, despite the fact that it is PLA, caught too much humidity even though I heard that PLA is not as susceptible to that, which could be one reason for the stringing. For now I will stick to the manual skirt object method and see what other things I can do to reduce the ooze (drying the filament, more bed calibration, deltas are a bit tricky from what I got, other filament brands etc.). And thanks for the input so far!