slic3r/Slic3r #4269: "Reload from Disk" does not work with network shares

dartrax, 2018-01-22:
### Version
19ea7a5

### Operating system type + version
Win 7 64bit

### Behavior
"reload from Disk" does simply nothing when the current stl model is located on a network share. Maybe "disk" is taken too literally... ;-)

### Steps to reproduce:
1. Save a .stl to network, something like: \\\\SATCAD01\Users\stl_3D-Druck\35052608.stl
2. Add to Slicer
3. Update .stl on network
4. Right click > Reload from Disk (nothing happens, not even status error message)

foreachthing, 2018-01-23:
Works on my Windows 7 64bit and Slic3r-master.2017.12.18.1561.d27b8a9.64bit
Added STL from `\\server\dir\dir\file.stl`, changed it, reloaded it in slic3r, then got new file.

dartrax, 2018-01-23:
Thank you @foreachthing for testing. I investigated further and the reason was that I splitted the model. I've updated the reproduce steps and the topic title, because the old one is misleading now.

foreachthing, 2018-01-23:
Jup! Now it does nothing :-)

lordofhyphens, 2018-01-23:
So we're expecting that after you make changes to the model in-memory (split is a change), hitting reload would discard everything and do a fresh load?

dartrax, 2018-01-23:
Yes.
When I have an updated model, every way to get this model into slicer leads to a point where I have to reiterate through all changes that I've made to this model, like positioning, rotation, cutting or splitting.
In fact, everything is already lost at the point where I decide that I have to update the model.

The perfect solution would probably to record all modifications that I've made to a model and to apply them automatically again after updating the model.

Positioning and rotations are being discarded as well right now and as end user I'd never make a difference between rotating the model and splitting it. I'm used to the fact that I lose positioning and rotating when reloading, so I expect to lose splitting and cutting, too. What I would not expect is that reloading doesn't work at all and does not give me a hint, why.