slic3r/Slic3r #4271: Feature request: Scale at Filament settings

hfuerst, 2018-01-23:
### Version
1.2.9

### Operating System
Win7-64bit

### Behavior
There should be a PreScaling (promille) in the "Filament Setting" Tab
because of material dependent shrinking after cool down.

For Example:
My Blue Filament shrinks 0,0098% after cooldown, 
so i need a preScale of 101,8% and all dimensions are ok.
But my Yellow Filament shrinks only 0,0073% - it is Material dependent!

Thx
HF

lordofhyphens, 2018-01-23:
Are you talking about XY size compensation?

timor, 2018-01-23:
It sounds to me as if this is exactly what the extrusion multiplier is for, no?

hfuerst, 2018-01-24:
No and No.

@lordofhyphens 
Mainly the shrinking is in XY but a little in Z too. Z is not as Much.

@timor 
I use the extrusion multiplier already. But this concerns only the size of the extruded material.
I mean the whole object(s). 
So far i use the Object Scale in the plater tab. But i have to keep in mind to do it, how much it is for each filament and it is not exactly enough. There are only Percent but it is necessary to have Promille.
-

hfuerst, 2018-01-24:
Sorry for the text size, I dont know how it came.

lordofhyphens, 2018-01-24:
Object scaling wouldn't work for what you seem to be asking for, as any shrinking is localized to islands. For small objects measurement error would mask any true differences. You would need to be printing single masses over half a meter (or larger) just to be able to reliably measure it.

For XY what you actually probably want is some voodoo math involving Xy size comp

lordofhyphens, 2018-01-24:
You had a trailing - on its own line in your comment. That was interpreted as "previous line is a header"