slic3r/Slic3r #4273: How to start printing from the actual lowest position of the stl?

powertide, 2018-01-24:
I've noticed that for a model to be sliced, slic3r puts it onto the print bed first, so that the gcode generated starts printing from the bed( i.e. from z=0). 
But now I want to get a preview of the printing from the lowest z position of the stl, instead of from z=0. I have found no such settings in slic3r GUI. Can anyone show me where to modify the source code to get it done? It'd be much appreciated.

lordofhyphens, 2018-01-24:
#4218

closed 2018-01-24 by lordofhyphens

powertide, 2018-01-25:
Thanks for your comment @lordofhyphens .
I found the closest possible answer to what I need in #4145 . But the problem remains unsoloved. 
To be clear, I am intending to get **Gcode** printing the object from the actual position of the stl, not only get the preview in GUI. I tried to remove the Z components of ModelObject::center_around_origin() and ModelObject::align_to_ground() in Model.cpp, as you suggested, and successfully recompiled. However, the gcode generated still touches the print bed.

lordofhyphens, 2018-01-25:
@GilesBathgate authored the PR

GilesBathgate, 2018-01-25:
@powertide Are you saying that the change in PR #4218 only affects the preview of the object, but not the actual output Gcode? That seems very strange to me. @lordofhyphens I am not really an expert with Perl, nor the Slic3r codebase and changed the code where I expected it needed to be. If the behaviour is still incorrect, then I might need some guidance to figure out the correct approach.

lordofhyphens, 2018-01-26:
@gilesbathgate turn on support with a fully suspended model.

powertide, 2018-01-26:
@GilesBathgate Yes. The code you changed seems to merely affect GUI, not the actual Gcode generation. Can you tell me where to change the code if I am to get what I want plz? @lordofhyphens
And I don't intend to use the support option because I want the Gcode generated by original model only.

lordofhyphens, 2018-01-26:
@powertide  Study the usage of https://github.com/alexrj/Slic3r/blob/master/xs/src/libslic3r/PrintObject.cpp#L171, that will probably get you most of the way there.

@GilesBathgate

powertide, 2018-01-30:
@lordofhyphens Thanks for your help. I changed the print_z and height in function "generate_object_layers" from PrintObject.cpp by adding the min z of the model to both of them. But Files such as "perimeters.t","skirt_brim.t","support.t" failed to compile. 
Where else should I alter? 
BTW is there a way to rebuild the altered files of the project, instead of having to wait forever for the "perl Build.pl" to finish?
@GilesBathgate  Have you gotten anywhere with this problem?

lordofhyphens, 2018-01-30:
If you configure local::lib to point at the right directory  (local-lib)
you can build in xs directly.