slic3r/Slic3r #4277: [feature request] different nozzle sizes for dual extruder for fast/strong+high resolution print

r0n0x, 2018-01-27:
Essentially, utilizing two nozzles of different sizes to acheive the benefits of both, which is high resolution and detail on smaller nozzle sizes, and faster but also stronger printing using larger nozzles. 

Lets say a combination of 0.2mm (lower end but still good for general use) and 1mm 

Here are some examples:
A wide fine mesh gear; this demands high detail on the teeth and accuracy on centering the axle/shaft hole in the center, but, for the bulk of the disk structure, it can be printed aggressively and benefits from the improved layer strength as well.

Large board with holes; lets say you have something normally better suited to being cut from sheet using a lasercutter. you want a board with specific notches and/or holes in it. with wide nozzles, holes often get filled in a fair bit, and this distorts the apparent center, making it difficult to correctly drill the center out. Also a very flat finish may be desired better suited to the finer nozzle.

There are clear benefits for some prints to have a different nozzle size used for certain parts. All prints can benefit from increased infill times, and even when theres no difference in print time, the strength improvement is undeniable. Using this for larger more invasive supports also would help. Especially when there is a very low layer height/ small nozzle being used.

In general though its the ability to place fine details that often should not need to be, or cannot be, machined into large otherwise basic and/or heavy prints after the fact. 

The only real weakness i can see for this is if/how the larger nozzle leaks when the small nozzle is printing, the inverse however is hardly an issue around infil, however extruder temp can be decreased accordingly since the nozzle difference will mean that, such as for the case of the 0.2mm and 1mm nozzle, assuming maximum layer height, the small nozzle must be used for 5 layers before the large one can be used 

Another feature to consider are making totally solid infill around small holes (good for medium/large gears which need good shaft grip, or when the hole still needs to be drilled out regardless of how fine its going to be, but centering is important.

lordofhyphens, 2018-01-27:
Um, you can already assign printing features to different extruders (infill, perimeters, etc).

What exactly are you asking for from Slic3r?

lordofhyphens, 2018-01-27:
Also see modifier meshes for changing the infill where you want . In the future one feature request per issue (there are 2 here).

7eggert, 2019-02-03:
Trying to translate: The user wants automatic layer height - but vertically by nozzle size: If the to-be-printed detail is less than x, use the fine nozzle; (maybe if it's the outer shell / {top,bottom} layer, too, as you suggested), otherwise use the coarse nozzle.