slic3r/Slic3r #4283: Hitting OK in DLP Projector Window produces crash.

N4rki, 2018-01-30:
TAGS: SVG, SLA

### Version
Version 1.3.0-dev  -   Slic3r-master.2018.01.21.1590.dc9d7ed.64bit
Also happens when building manually: 1.2.9-1272-gdc9d7ed7

Windows 10

### Behavior
DLP Slicing window produces crash immeadiately after hitting OK.
Reproduce: 
- Load any file into slic3r (I tried differnt .stl and .3mf files
- Go to Window -> DLP Projector and hit OK
- I tried many different setting like 0% or 100% infill and rectangual infill with the same results
Expected reesults:
Never seen it working but probably the slicing would finish succesfully.
Actual results: 
A small loading window pops up, saying "processing layers", with black canvas and red raster pattern in background. Before loading bar can reach more than a few percent, another window pops up, saying "3D Printer Application has stopped working".

That last windows gives me the option to either close the apllication or debug it. If I hit debug, Visual Studio tells me the following:
"Ausnahme ausgelöst bei 0x000000002B92BB27 (XS.xs.dll) in Slic3r.exe: 0xC0000094: Integer division by zero."
First three words can be translated to: Exception triggered at...

#### STL/Config (.ZIP) where problem occurs
Any will do: [https://github.com/alexrj/Slic3r/files/1343013/4130.zip]
[config.zip](https://github.com/alexrj/Slic3r/files/1679260/config.zip)

Any ideas? I have read it somewhat worked for other people, so I'm assuming this isn't just not yet implemented.

lordofhyphens, 2018-01-31:
In my own experience it is only currently working at 100% infill. I haven't
had time to look into it further.

N4rki, 2018-02-19:
I just found out that the 100% infill has to set from the main settings, not from the DLP Projector Window in order to circumvent the crash. This is as described in #4238.

lordofhyphens, 2018-02-25:
@N4rki it's actually because the auto width for infill extrusion width is "0", which happens to be used in SLAPrint for min_spacing. I've submitted a hotfix for that (set  the infill extrusion width to 0.3 if 0 in config).

lordofhyphens, 2018-02-25:
I'll look into a nicer way of fixing this.

lordofhyphens, 2018-02-25:
Found a nicer way to fix it which was to actually make SLAPrintOptions do something (it wasn't before).

closed 2018-02-25 by lordofhyphens