slic3r/Slic3r #4286: Feature request: Change export g-code behaviour with temp files

MiggyMan, 2018-02-01:
Slightly long title but I couldn't think of a better way to say it :D

A little background on the request, I use octoprint's watch folder functionality via a samba exported folder, I noticed that any time I saved a file into it via slic3r it wasn't picked up but with any other slicer it was, it took me a while to spot it but I noticed that slic3r saves to a temporary file first before renaming the gcode to it's final name, this seems to throw the watch functionality, I suspect because renaming a temp file wouldn't register as a new file and when it is first opened it doesn't have the appropriate filename extension!

So, to that end would it be possible to have the behaviour of the export button changed to save the temporary file to a given path and then move it over to the resulting gcode file to it's final destination ?

foreachthing, 2018-02-02:
You could write your own batch-script and use the post-processing (Print Settings - Output options - Post processing script).
Kind of: copy A to B then delete A.

MiggyMan, 2018-03-07:
Closing, seems a non operating system agnostic workaround is the only option to this tiny amendment.

closed 2018-03-07 by MiggyMan