slic3r/Slic3r #4287: [feature request] Shut down bed heating after certain height

probonopd, 2018-02-02:
It would be very useful to be able to switch off the bed heating after height X.

Rationale: After a sufficient number of layers, the heating is not needed anymore (depending on the material and model). Being able to switch off (or turn down) the heating after a sufficient number of layers would save the time it takes to cool down the bed at the end of the print before one can remove the parts from the bed. (At least with my PLA prints, I do this on a regular basis with great success.)

bubnikv, 2018-02-02:
You can do that in Slic3r with the conditional G-code syntax inside the
layer change G-code, based on the current layer height.

probonopd, 2018-02-02:
> You can do that in Slic3r with the conditional G-code syntax inside the layer change G-code, based on the current layer height.

Wow, actually I was looking for something like that but only found boxes for entering "Start" and "End" G-code. Where can I enter "conditional G-code syntax inside the layer change G-code"?

bubnikv, 2018-02-02:
https://github.com/alexrj/Slic3r/wiki/Conditional-Gcode-Syntax-Spec

[image]

The Prusa3D Slic3r implements a bit different conditional syntax 
{if condition} xxx {else} xxxx {endif}
and we started to use the conditional syntax for things like controlling the extrusion multiplier based on the layer height. Just search the "{if" string inside our Slic3r profiles.

probonopd, 2018-02-02:
Thanks @bubnikv - I can see commands like `{if not has_wipe_tower}` and `{if printer_notes=~/.*PRINTER_HAS_BOWDEN.*/}` but how do I specify "if at the first layer at or above X cm" or "if at layer number Y"?

bubnikv, 2018-02-02:
You enter the following into the Before layer change G-code section to disable bed heater above 10mm:

If it is the upstream Slic3r, then you will enter:
{if [layer_z] > 10}M140 S0

If it is the Slic3r PE, you will enter
{if layer_z > 10}M140 S0{endif}

Actually this will enter the M140 S0 at each layer above 10mm. You may be a bit more creative by limiting the upper limit, with the Slic3r PE syntax:
{if layer_z > 10 and layer_z < 11}M140 S0{endif}

probonopd, 2018-02-02:
Thanks a lot @bubnikv. This will work for me.

If others think this is a useful feature too, maybe it can be made a GUI element ("switch off heating after height __ cm)."

probonopd, 2018-03-25:
After having used this for a while, I really think it may be worthwhile to have a GUI element ("switch off heating after height __ cm") to make this more accessible for everyone.